espressif/esp-matter #663: mfg_tool.py giving error

bilalmalik76, 2023-09-28:
I am trying to use mfg_tool to generate partition binary but  I am getting this error: ModuleNotFoundError: No module named 'esp_secure_cert'

I  have also installed requirements.txt
[image]

shubhamdp, 2023-09-29:
@bilalmalik76 I think of two possibilities

I think you are using the older esp-matter SHA or the requirements are not installed in esp-idf's python environment.

First, you may be using older esp-matter revision, and that was fixed in https://github.com/espressif/esp-matter/commit/4ebc133745d4e7f8b297ae7e47c8fe9795cb0756, try the latest esp-matter.

For second, requirements may not be installed in esp-idf's python environment, try below steps for that.
```
cd esp-matter/tools/mfg_tool
source esp-idf/export.sh
python3 -m pip install -r requirements.txt
```

[Install python dependencies](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool/#install-python-dependencies) section does not mention the sourcing of idf environment and will fix that in the doc.

bilalmalik76, 2023-09-29:
@shubhamdp Thanks for your response. This does not fix it. So I tried to install the latest esp_matter but now getting this error:
[image]

shubhamdp, 2023-09-29:
Can you try again by removing the connectedhomeip environment.
```
rm -rf connectedhomeip/connectedhomeip/.environment
source path/to/esp-idf/export.sh
./install.sh
```

shubhamdp, 2023-09-29:
Also, rather than pasting a screenshot of error, complete logs from the run would help to debug problem faster.

bilalmalik76, 2023-09-29:
@shubhamdp still getting the same error after doing above mentioned steps. Please see the logs: 
`bilal@Ubuntu22:~/Desktop/esp-matter$ rm -rf connectedhomeip/connectedhomeip/.environment

bilal@Ubuntu22:~/Desktop/esp-matter$ source /

bin/        cdrom/      etc/        lib/        lib64/      lost+found/ mnt/        proc/       run/        snap/       swapfile    tmp/        var/        

boot/       dev/        home/       lib32/      libx32/     media/      opt/        root/       sbin/       srv/        sys/        usr/        

bilal@Ubuntu22:~/Desktop/esp-matter$ source /

bin/        cdrom/      etc/        lib/        lib64/      lost+found/ mnt/        proc/       run/        snap/       swapfile    tmp/        var/        

boot/       dev/        home/       lib32/      libx32/     media/      opt/        root/       sbin/       srv/        sys/        usr/        

bilal@Ubuntu22:~/Desktop/esp-matter$ source ~/

.bash_history                                .gitconfig                                   .sudo_as_admin_successful                    .vboxclient-seamless-tty2-service.pid

.bash_logout                                 .gnupg/                                      Templates/                                   .vboxclient-vmsvga-session-tty2-control.pid

.bashrc                                      .local/                                      temp-sensor/                                 .vboxclient-vmsvga-session-tty2-service.pid

.cache/                                      Music/                                       TEST_CD_FFF1_8001.der                        Videos/

.cipd-cache-dir/                             .pam_environment                             .vboxclient-clipboard-tty2-control.pid       .vscode-server/

.config/                                     Pictures/                                    .vboxclient-clipboard-tty2-service.pid       .wget-hsts

Desktop/                                     .profile                                     .vboxclient-draganddrop-tty2-control.pid     .Xauthority

Documents/                                   Public/                                      .vboxclient-draganddrop-tty2-service.pid     

Downloads/                                   snap/                                        .vboxclient-hostversion-tty2-control.pid     

.espressif/                                  .ssh/                                        .vboxclient-seamless-tty2-control.pid        

bilal@Ubuntu22:~/Desktop/esp-matter$ source ~/

.bash_history                                .gitconfig                                   .sudo_as_admin_successful                    .vboxclient-seamless-tty2-service.pid

.bash_logout                                 .gnupg/                                      Templates/                                   .vboxclient-vmsvga-session-tty2-control.pid

.bashrc                                      .local/                                      temp-sensor/                                 .vboxclient-vmsvga-session-tty2-service.pid

.cache/                                      Music/                                       TEST_CD_FFF1_8001.der                        Videos/

.cipd-cache-dir/                             .pam_environment                             .vboxclient-clipboard-tty2-control.pid       .vscode-server/

.config/                                     Pictures/                                    .vboxclient-clipboard-tty2-service.pid       .wget-hsts

Desktop/                                     .profile                                     .vboxclient-draganddrop-tty2-control.pid     .Xauthority

Documents/                                   Public/                                      .vboxclient-draganddrop-tty2-service.pid     

Downloads/                                   snap/                                        .vboxclient-hostversion-tty2-control.pid     

.espressif/                                  .ssh/                                        .vboxclient-seamless-tty2-control.pid        

bilal@Ubuntu22:~/Desktop/esp-matter$ source ~/

.bash_history                                .gitconfig                                   .sudo_as_admin_successful                    .vboxclient-seamless-tty2-service.pid

.bash_logout                                 .gnupg/                                      Templates/                                   .vboxclient-vmsvga-session-tty2-control.pid

.bashrc                                      .local/                                      temp-sensor/                                 .vboxclient-vmsvga-session-tty2-service.pid

.cache/                                      Music/                                       TEST_CD_FFF1_8001.der                        Videos/

.cipd-cache-dir/                             .pam_environment                             .vboxclient-clipboard-tty2-control.pid       .vscode-server/
[... 1369 lines cut ...]
  File "/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 682, in virtualenv

    if not virtualenv_setup.install(

  File "/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 296, in install

    pip_install(

[... cut ...]
subprocess.CalledProcessError: Command '['/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/bin/python', '-m', 'pip', 'install', '--log', '/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/pip-requirements.log', '--requirement=/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/scripts/setup/requirements.txt', '--constraint=/home/bilal/Desktop/esp-matter/connectedhomeip/connectedhomeip/scripts/setup/constraints.txt']' returned non-zero exit status 1.

bilal@Ubuntu22:~/Desktop/esp-matter$ 

`

shripad621git, 2023-10-12:
@bilalmalik76, please provide the error logs of the mfg_tool in a file.

dhrishi, 2023-12-01:
@bilalmalik76 Please close this issue if resolved

closed 2023-12-04 by dhrishi