espressif/esp-matter #666: Compilation of Matter examples fail in VSCode on Ubuntu

baylf2000, 2023-10-02:
**Describe the bug**
Attempting to compile any Matter examples in VSCode on Ubuntu 23.04 fail with error:
`ERROR at //build_overrides/pigweed_environment.gni:20:1: Unable to load ... pigweed_environment.gni`

**Environment**
- ESP-Matter Commit Id: 1cacb8b2fcd42c465e38317d3a63d0169460b397
- ESP-IDF Commit Id: cb174b0fe1f150f2ecc2ce87e0f02154405c6677
- SoC (eg: ESP32 or ESP32-C3): ESP32 or ESP32-C6
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 23.04

Complete compile log attached.

**Any additional details**
...
[esp_matter_compile.txt](https://github.com/espressif/esp-matter/files/12782438/esp_matter_compile.txt)

jonsmirl, 2023-10-02:
Did you run install.sh in the top directory?

baylf2000, 2023-10-03:
> Did you run install.sh in the top directory?

I had done so, but I did again and it has corrected the error. Thank you!

closed 2023-10-03 by baylf2000