espressif/esp-matter #667: Compilation fails on the ESP32-C6

baylf2000, 2023-10-03:
**Describe the bug**
When the target is set as `esp32c6` the compilation immediately fails with the following error:

```
-- Found Git: /usr/bin/git (found version "2.39.2") 
CMake Error at /home/lachlan/esp/esp-matter/device_hal/device/esp32_devkit_c/esp_matter_device.cmake:3 (message):
  please set esp32 as the IDF_TARGET using 'idf.py set-target esp32'
Call Stack (most recent call first):
  CMakeLists.txt:35 (include)

-- Configuring incomplete, errors occurred!
  ```

**Environment**
- ESP-Matter Commit Id: 1cacb8b2fcd42c465e38317d3a63d0169460b397
- ESP-IDF Commit Id: cb174b0fe1f150f2ecc2ce87e0f02154405c6677
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Host Machine OS: Ubuntu 23.04

shubhamdp, 2023-10-03:
@baylf2000 Which example are you trying out? Not all the examples have the below line which adds device path https://github.com/espressif/esp-matter/blob/1cacb8b2fcd42c465e38317d3a63d0169460b397/examples/light/CMakeLists.txt#L20-L22

Can you check the example that you are using have this set correctly?

baylf2000, 2023-10-03:
> @baylf2000 Which example are you trying out? Not all the examples have the below line which adds device path [...]
I'm currently using the "Light_Switch" example. Yes it does have the lines you posted above, but it does not work. I have to manually export `ESP_MATTER_DEVICE_PATH` otherwise it won't compile.

baylf2000, 2023-10-03:
I've just tried building from the command line provided by "Open ESP-IDF Terminal", and it works, so the issue must be with VSCode somehow.

sayondeep, 2023-10-03:
@baylf2000 you can set the ESP_MATTER_DEVICE_PATH inside vscode using command palette (Ctrl + Shift + P ). `ESP-IDF:Set ESP-MATTER Device Path (ESP_MATTER_DEVICE_PATH)`. Once you have set the path appropriately the build should work.

baylf2000, 2023-10-06:
Thank you @sayondeep . This does solve the problem.

closed 2023-10-06 by baylf2000