espressif/esp-matter #669: BLE-WiFi commissioning fails for ESP32 with PSRAM

jzwyssig, 2023-10-03:
BLE-WiFi pairing fails when PSRAM is enabled. The same settings on a ESP32-S3 work perfectly fine. Does anyone have an idea what to try? MbedTLS, esp-matter and Bluetooth are set to allocate memory in PSRAM. Our firmware also overflows IRAM on ESP32 but not on ESP32-S3. That is why I enabled [CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-system-esp32-sram1-region-as-iram) which solved the issue. Everything else is working fine, except for commissioning.

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/0a1d89d7235a5dc632edd488727814b9690b6270 (up to date)
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986 (5.1.1 stable)
- SoC: ESP32
- Host Machine OS: Linux
- Sdkconfig: [sdkconfig.esp32.txt](https://github.com/espressif/esp-matter/files/12795209/sdkconfig.esp32.txt)
- Logs of the device are not helpful

[image]

shubhamdp, 2023-10-04:
@jzwyssig Rather than a screenshot the complete logs from the device in a file attached here would be more helpful.

On commissioner side, which commissioner are you using? If you are using chip-tool then logs from the chip-tool would be more helpful.

jzwyssig, 2023-10-04:
@shubhamdp I attached the log file.

mgoeller, 2023-10-05:
I get the same error with an ESP32

jonsmirl, 2023-10-05:
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/12819761/sdkconfig.txt)

I am using this SDK config with the light example on ESP32+PSRAM without BLE problems.

jonsmirl, 2023-10-11:
BTW, your commissioning worked. It is the wifi-based mDNS advertising which is failing.

jzwyssig, 2023-11-16:
I close this for now. This issue is not our priority and I did not investigate it. I will reopen it as necessary.

closed 2023-11-16 by jzwyssig