espressif/esp-matter #670: Request for a level measurement example

MichaelHeiser, 2023-10-04:
**Is your feature request related to a problem? Please describe.**
The device which should integrate into a matter environment measures the level of a water or oil tank for example. It seems that with cluster level_control you can only set levels to a specific value and not to measure the level and send the level to the controller for making an alarm for example.

**Describe the solution you'd like**
- Device measures the level of a liquid / gas / solid quantity
- there is a max level and a min level defined
- on reaching min level an alert has to be notified (eg. a siren via a contact, notification to smartphone)
- quantity needs to have a name ("water")
- level needs to have a unit (l or l/m³) 

**Describe alternatives you've considered**
The light example is technically the same but with setting the on/off and level adjustments and not notifying these quanitities

shubhamdp, 2023-10-04:
@MichaelHeiser why don't you subscribe to  the desired attribute on your client/controller and take the required action.

MichaelHeiser, 2023-10-04:
For clarification, I am really new to Matter and it would also be really nice if I will be pointed into the correct direction. Can you give me advice what you mean witd subscribing to the attribute. Is it comparable with an indication to a characteristic in BLE? Can I control the write rights to an attribute? Also a pointing to the correct doc section is welcome.

Thanks in advice :)

shubhamdp, 2023-10-05:
Please find below few resources
- [The ESP Journal](https://blog.espressif.com/matter-38ccf1d60bcd) blogs for high level understanding of Matter.
- [Commissioning and Control](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) section from ESP Matter Programming Guide has some details around subscription.
- And Matter specification can be found at [CSA's official site](https://csa-iot.org/developer-resource/specifications-download-request/)

shubhamdp, 2023-10-19:
@MichaelHeiser Do you have any update on this or if your question is answered please close the issue.

MichaelHeiser, 2023-10-19:
Can you point me to a real code example, where the attributes are just read from a matter device?

dhrishi, 2023-10-21:
> Can you point me to a real code example, where the attributes are just read from a matter device?

Reads, when issued by a commissioner, are taken care of internally by the core Matter SDK. The code to read OnOff attribute: https://github.com/espressif/connectedhomeip/blob/699ff65e065a28e4b7ee3a2d84110444b1709040/src/app/clusters/on-off-server/on-off-server.cpp#L421

Reading directly on the device, in esp-matter, is done like [this](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#L62-L68)

dhrishi, 2023-10-30:
Closing as the questions are answered

closed 2023-10-30 by dhrishi