espressif/esp-matter #671: Bridge and matter device communitation

mttrRDRSDX, 2023-10-05:
**Describe the question/query that you have**

as i can see there is a example 
[esp-now_bridge_light](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light)

in this example we are doing some binding of the bridge to the matter esp light
will it help in
binding the bridged device to matter device ?
like if i toggle the bridged device then matter device light will glow 

**Additional context**
...

bubblesnake, 2023-10-06:
Hi @mttrRDRSDX , binding the bridge with a Matter light is optional. It allows the bridged device to control other Matter devices via binding. Without binding, it is still possible to send onoff commands via direct addressing.

The bridge has a dynamic endpoint of OnOff switch device type. When it binds with an ESP-NOW initiator, the dynamic endpoint is created. Then the bridge works just like a normal OnOff switch device. Hence the instruction of creating the binding is similar to the light switch example. You can see in the [esp-now_bridge_light](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light) README that the endpoint 3 has a binding server cluster.

mttrRDRSDX, 2023-10-07:
**Without binding, it is still possible to send onoff commands via direct addressing.**

without binding also but how can bridged device or bridge  discover the matter device ?

so this ESP-NOW <-> ESP Initiator is like a attribute translator?
so ESP-Initiater will do some toggling and corresponding to that mapped matter device attribute will toggle right ?
in case of onoff example
thats how esp-intiator - responder working 
making it attribute translator

bubblesnake, 2023-10-07:
**without binding also but how can bridged device or bridge discover the matter device ?**

Without binding, you will need to find a way to get the matter device's address. It would be the same use case as the light switch example.

You can think of a matter bridge as a translator between a communication technology and the matter.

mttrRDRSDX, 2023-10-08:
Finding a matter device is not difficult right 
we can use dns-sd discovery 
matter._tcp_ something 
but i was trying to extend this bridge example 
and i was stuck at part on how this node address can be discovered 
because it would good right if this bridge can be extended to any matter device then only matter light app .

so can you bit help on finding this answere 
if i discovered a matter app with the dns discovery or say avahi
then how this node address or id can be used to send command to it ?
without binding .
i hope its possible

dhrishi, 2023-10-09:
@mttrRDRSDX 

Performing any kind of actions to control a Matter node requires:
1. Updating the address of the destination Matter node, on the source node
2. Updating the access control privileges to add the entry of source node, on the destination node

Now, 1. above can be considered as updating the binding attribute of the binding cluster on the source node. If you know the address (node id) of the Matter node you want to communicate to then, please confirm of the controlling node has the access to control. If not, add/update the same.

The last part of sending the read/write/invoke commands to the other node are the same. You can refer [this](https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/app_driver.cpp#L102-#L124)

mttrRDRSDX, 2023-10-09:
i see if the proper ACL is there 
then we can invoking client::connect -> which further invokes     server->GetCASESessionManager()->FindOrEstablishSession(ScopedNodeId(node_id, fabric_index), &success_callback,
                                                            &failure_callback);
to establish a connection with the matter device right . now i understood the process i missed client::connect previously.

seems helpful and what i was looking for thanks for now 
you can close the ticket 
if anything pops up i will continue in this ticket is that fine ?

dhrishi, 2023-10-09:
@mttrRDRSDX Yes, sure.

closed 2023-10-09 by dhrishi

mttrRDRSDX, 2023-10-13:
i Have a query i am able to extend the bridge as discussed but one issue is 
i every time i want to send command to a node i need to connect via FindOrEstablishSession then am able to send the commands
but when am trying to directly send the command after establishing session lets say 
after some time 
then i get a crash 

is it an expected behaviour that every time we need to FindOrEstablishSession before sending any command to node ?

mttrRDRSDX, 2023-10-16:
Any update on this query ?