espressif/esp-matter #672: Commissioning unreliable on ESP32-C6 (via WiFi)

baylf2000, 2023-10-06:
**Describe the bug**
Commissioning on an ESP32-C6 is very unreliable. Fails 4 out of 5 times, and the 5th time requires several "retry" attempts on the commissioning device. Using the "Light" example.

**Environment**
- ESP-Matter Commit Id: 1cacb8b2fcd42c465e38317d3a63d0169460b397
- ESP-IDF Commit Id: cb174b0fe1f150f2ecc2ce87e0f02154405c6677
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file): Device log file attached.
- Host Machine OS: Linux (Ubuntu 23.04)
- Commissioner app and versions if present: Home Assistant Android App (Version 2023.9.2-full)
- Commissioner's logs if present:

**Any additional details**
...
[ESP32C6_Matter_Log_02.txt](https://github.com/espressif/esp-matter/files/12828372/ESP32C6_Matter_Log_02.txt)

baylf2000, 2023-10-06:
Just for further context, I am able to build and flash the same "Light" example on an ESP32-S3 and commission it without problems, most of the time.

dhrishi, 2023-10-06:
@baylf2000 Just to confirm, can you flash the light example from Launchpad [here](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml) and check if you see the same behavior. It is Matter over Wi-Fi (light example) binary, for C6.

baylf2000, 2023-10-07:
> @baylf2000 Just to confirm, can you flash the light example from Launchpad [here](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml) and check if you see the same behavior. It is Matter over Wi-Fi (light example) binary, for C6.

@dhrishi That firmware worked first time, no problems. So I guess the question is, what is different between that and the example firmware in the sdk?

baylf2000, 2023-10-10:
@dhrishi Any idea where I should start here? I tried again today, compiling the light example without any changes and it has the same commissioning issues as above, yet the compiled example you provided works perfectly. I'd really like to know what is different between your compiled version and the example in the sdk.

dhrishi, 2023-10-10:
@baylf2000 Ideally they should be the same. Although, as I check, the current launchpad binaries are around a month back and haven't updated due some setup related issues there. We will check with the latest codebase on ESP32-C6 and if that fails, it is a regression on our side and will be fixed. Please allow us some time to figure this out.

VaishaliAvhale, 2023-10-12:
@baylf2000, I have tried commissioning the ESP32-C6 using BLE-WiFi with the latest ESP-Matter commit, and it is working well on my side.
- Please check whether your pairing network matches with your system's because it should be the same.
- If it still doesn't work, please attach the chip-tool and device log files.

baylf2000, 2023-10-16:
@dhrishi @VaishaliAvhale I pulled the recent updates and tried again and it appears to be working fairly reliably now, although commissioning seems to take much longer than with other, commercially released devices.

I'll close this issue for now though. Many thanks for the assistance.

closed 2023-10-16 by baylf2000