espressif/esp-matter #674: light-demo has slow memory leak

jonsmirl, 2023-10-08:
I boot light demo let it stabilize with a wildcard report monitoring it.
~~~
Totals:
    free 81128 allocated 148396 min_free 63104 largest_free_block 65536
~~~
then five minutes later
~~~
  Totals:
    free 81132 allocated 148392 min_free 63104 largest_free_block 65536
~~~
i took the wifi router up and down
~~~
  Totals:
    free 81528 allocated 148028 min_free 63104 largest_free_block 65536
~~~
then I let it sit for two days. During those two days the monitoring wild card lost contact and then reestablished it several times. The wildcard app is on my phone, so when I went out of range of the router it lost contact.
~~~
  Totals:
    free 76084 allocated 153076 min_free 18780 largest_free_block 65536
~~~

It is slowly allocating more and more memory so something is leaking. I went from 148396 to 153076 over the course of two days. So I have lost 4680 bytes of memory. I think this is behind the packet buffer allocation failures.

I attached the change I made to light demo and the sdkconfig. 
[sdkconfig.txt](https://github.com/espressif/esp-matter/files/12840554/sdkconfig.txt)
[light.diff.txt](https://github.com/espressif/esp-matter/files/12840557/light.diff.txt)
esp-matter 1cacb8b2fcd42c465e38317d3a63d0169460b397
I am using an esp32 dev board with PSRAM
esp-idf e088c3766ba440e72268b458a68f27b6e7d63986

jonsmirl, 2023-10-08:
In another experiment I turned on the leak tracking support in the IDF. I could not figure out how to get useful data out of it because I don't have a good place to start/stop the data collection. 
[leak.txt](https://github.com/espressif/esp-matter/files/12840580/leak.txt)
That trace is from my code which also has BT turned on. The start/stop was done about three hours apart.

You can see what the problem is. I can't tell allocations which are two hours old from transient allocations which will be less than a second old.  Many of those entries are not leaks, they are transient allocations coming from other tasks in the system like wifi. Those transient allocations just happened to be active when I did STOP on the leak detector.  The dump info needs to include the age of the allocation. Once you have the age, the oldest ones will be the likely leaks. A two hour age will be a leak, a 1 second age transient.

jonsmirl, 2023-10-08:
BTW, I don't think (not 100% sure) it leaks just sitting there. The leaks appear to be something either with the router disconnect/reconnect or the wild card monitor disconnect/reconnect.

jonsmirl, 2023-10-11:
This could be a potential leak source: https://github.com/espressif/esp-matter/issues/656

I have not added enough debug tracking to know.

jonsmirl, 2023-10-22:
I had light example running again for 3.9 days. I need to reboot it to change config. I am running light example using common ESP32 dev boards since I have a lot of those.

Start 146,852 allocated
End 153,924 allocated

7,024 lost. For sure there is noise in these numbers due to wifi activity. Previous run lost 7,952 in 3.6 days. Previous run was likely interrupted when it had some wifi activity going on.  But there is a definite trend line due to leaks which keeps going up. 

Using...
~~~
commit c707e6680ac39df56b50addfe6792d49a60f2f90 (origin/main, origin/HEAD, main)
Merge: bbdf797 9e53da8
Author: Shu Chen <chenshu@espressif.com>
Date:   Wed Oct 18 16:10:33 2023 +0800
~~~

shubhamdp, 2023-11-06:
@jonsmirl  We are narrowing down the source of the leak

jonsmirl, 2024-05-28:
I am not seeing leaks in esp-matter anymore.

closed 2024-05-28 by jonsmirl