espressif/esp-matter #675: Which application is the best for testing matter functionality

damian-kurek-wizzdev, 2023-10-09:
**Describe the question/query that you have**
I want to test the matter thermostat device I created using esp-matter SDK. Which mobile app is the best for this purpose? I tried Google Home, but I can only set the temperature and mode (I do not support cooling, but the app still shows it as a possibility). (I can create routines etc. But they are controlled by Google Home and not transferred to the device)Is there an app that supports more matter endpoints like weekly schedules?  
**Additional context**
...

jonsmirl, 2023-10-11:
IOS would be the only other option and I have not tried thermostat on it.
chip-tool will work from the command line.

If those fail, you can always write your own app.

damian-kurek-wizzdev, 2023-10-16:
[image]
I found out why, scheduling is not implemented on connectedhomeip side. I guess I need to wait with it

closed 2023-10-16 by damian-kurek-wizzdev

jonsmirl, 2023-10-16:
They can't be implemented in the common code because they need to know the time and CHIP doesn't not have TimeSource implemented yet.  But you can implement them yourself if you have NTP active.  I believe they are marked for weak linkage so if you redefine in your app you can override.

damian-kurek-wizzdev, 2023-10-16:
They aren't marked as weak functions. Shouldn't this function at least implement parsing of the schedule?
It doesn't make sense to write a comment TODO when a user needs to implement it.

jonsmirl, 2023-10-16:
Make a stub in your code and see if it will link

damian-kurek-wizzdev, 2023-10-17:
I checked it, It does not compile "multiple definitions of".

jonsmirl, 2023-10-17:
Since it is your own local copy, just comment the stub out in CHIP. Then you can work on your code.