espressif/esp-matter #676: should remove wifi credentials after remove all fabrics?

nykh2010, 2023-10-10:
https://github.com/espressif/esp-matter/blob/63e79f02114cc795ad9e4c68303076bb89cac925/examples/light/main/app_main.cpp#L84C21-L86C24

If not remove wifi credentials after remove all fabrics, device cannot change wifi credentials in new commission, because in AddOrUpdateNetwork function will match incomming ssid with mStagingNetwork, then the result is return Status::kBoundsExceeded.

wqx6, 2023-10-12:
After removing all fabrics, the device will open its commissioning window and allow other commissioner to commission it to a new fabric. If you want to switch W-Fi credentials after commissioning, you can use `RemoveNetwork` `AddOrUpdateNetwork`, and `ConnectNetwork` command in the `NetworkCommissioning` cluster.

PSONALl, 2023-10-12:
@nykh2010 According to spec all the matter data should be removed after last fabric removal,
For device which runs Matter and some other protocol in parallel. Deleting network connectivity could have unwanted side effects.

That's why in this use case if you want to remove the wifi credentials you can add below lines to your code when you recieve kFabricRemoved Event  :
```
    case chip::DeviceLayer::DeviceEventType::kFabricRemoved:
        {
            ESP_LOGI(TAG, "Fabric removed successfully");
            if (chip::Server::GetInstance().GetFabricTable().FabricCount() == 0)
            {
                chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
                constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(k_timeout_seconds);
                if (!commissionMgr.IsCommissioningWindowOpen())
                {
                    // Initialise BLE manager
                    CHIP_ERROR err = chip::DeviceLayer::Internal::BLEMgr().Init();
                    if (err != CHIP_NO_ERROR) {
                           ChipLogError(DeviceLayer, "BLEManager initialization failed: %s", ErrorStr(err));
                    }  
                    // Clear Wifi credentials
		    chip::DeviceLayer::ConnectivityMgr().ClearWiFiStationProvision();
		    // Advertise over BLE
                    chip::DeviceLayer::ConnectivityMgr().SetBLEAdvertisingEnabled(true);
		    err = commissionMgr.OpenBasicCommissioningWindow();
                    if (err != CHIP_NO_ERROR)
                    {
                        ESP_LOGE(TAG, "Failed to open commissioning window, err:%" CHIP_ERROR_FORMAT, err.Format());
                    }
                }
            }
```

closed 2023-10-13 by nykh2010