espressif/esp-matter #681: Scenes Cluster on a dynamic endpoint doesn't initialize.

jonsmirl, 2023-10-12:
My fixed endpoints have scenes clusters on them so the scene server is initialized. Then I create a dynamic endpoint in my bridge which also contains a scene cluster. It errors out because the scenes server is already initialized.
~~~
I (281066) esp_matter_bridge: Cluster plugin init for the new added endpoint
E (281076) chip[ZCL]: Duplicate attribute override registration failed
E (281076) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (281076) esp_matter_core: Error adding dynamic endpoint 3: 0x8a
~~~
Error 3 is 'incorrect state'
~~~
CHIP_ERROR ScenesServer::Init()
{
    // Prevents re-initializing
    VerifyOrReturnError(!mIsInitialized, CHIP_ERROR_INCORRECT_STATE);
~~~

wqx6, 2023-10-13:
The first two error lines are expected and could be ignored as the plugin_server_init should only be called once, but the error `E (281076) esp_matter_core: Error adding dynamic endpoint 3: 0x8a` is strange as the `0x8a` is `EMBER_ZCL_STATUS_DUPLICATE_EXISTS` which will only be returned when the  endpoint ID is used. Is the endpoint_id 3 used before?

jonsmirl, 2023-10-13:
Not sure what was causing 0x8a. When I erased the flash it went away so I can't debug it.

Now I can make the endpoint. It gets the error 3 but it continues on.
~~~
I (1911) ep: JDS98 bridge_device_type_callback_t dt fff30003 group is 0
I (1911) esp_matter_bridge: Cluster plugin init for the new added endpoint
E (1911) chip[ZCL]: Duplicate attribute override registration failed
E (1911) chip[ZCL]: ScenesServer::Instance().Init() error: 3
I (1911) chip[DL]: Updating advertising data
I (1911) esp_matter_attribute: ********** W : Endpoi
~~~

closed 2023-10-13 by jonsmirl