espressif/esp-matter #684: Intellisense fail - Ubuntu & VSCode

baylf2000, 2023-10-16:
Intellisense is largely broken when using Matter examples in VSCode on Ubuntu.

It seems to be unable to recognise files or their contents from Matter or the current project. The project builds correctly though.

**Environment**
- ESP-Matter Commit Id: 79dfcc7b9db37d80105e2bbb39a680a0670296d5
- ESP-IDF Commit Id: c1c843f5e2315ba050e13a9e0738ee8abd506827
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Host Machine OS: Ubuntu 23.04

Intellisense errors from "Light" example project:

```
[{
	"resource": "/home/lf/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"device.h\"",
	"source": "C/C++",
	"startLineNumber": 13,
	"startColumn": 1,
	"endLineNumber": 13,
	"endColumn": 20
},{
	"resource": "/home/lf/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"esp_matter.h\"",
	"source": "C/C++",
	"startLineNumber": 14,
	"startColumn": 1,
	"endLineNumber": 14,
	"endColumn": 24
},{
	"resource": "/home/lf/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"led_driver.h\"",
	"source": "C/C++",
	"startLineNumber": 15,
	"startColumn": 1,
	"endLineNumber": 15,
	"endColumn": 24
},{
	"resource": "/home/lf/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "#include errors detected. Consider updating your compile_commands.json or includePath. Squiggles are disabled for this translation unit (/home/lachlan/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp).",
	"source": "C/C++",
	"startLineNumber": 17,
	"startColumn": 1,
	"endLineNumber": 17,
	"endColumn": 22
},{
	"resource": "/home/lf/Documents/esp_projects/MATTER_PROJECTS/light/main/app_driver.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "cannot open source file \"esp_matter.h\" (dependency of \"app_priv.h\")",
	"source": "C/C++",
	"startLineNumber": 17,
	"startColumn": 1,
	"endLineNumber": 17,
	"endColumn": 22
}]
```

jonsmirl, 2023-10-16:
c-cpp-properties
~~~
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${default}",
                "/home/jonsmirl/aosp/esp-matter"
            ],
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
~~~

baylf2000, 2023-10-16:
@jonsmirl Thank you. That fixes it! Perhaps the maintainers could add that fix to the example projects?

jonsmirl, 2023-10-16:
There is VSCode command called "ESP-IDF: Add vscode configuration folder"  but what I sent you will work better.

bamnet, 2023-11-23:
FWIW I ended up just adding `"${config:idf.espMatterPath}/**",` as a line, resulting in the following config --

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "${config:idf.toolsPath}/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${config:idf.espMatterPath}/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${config:idf.espMatterPath}/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

```

sayondeep, 2023-11-23:
@baylf2000 can you please close the issue if it is resolved.

baylf2000, 2023-11-24:
@sayondeep No this issue is not resolved. The maintainers need to fix the example projects. There are several possible solutions described above, but until the source code is corrected the issue will remain.

sayondeep, 2023-12-11:
@baylf2000 vs code configuration related information is available [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md). You can use them for your reference and configure vs code.

khuee, 2025-03-02:
> [@baylf2000](https://github.com/baylf2000) vs code configuration related information is available [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md). You can use them for your reference and configure vs code.

the link  died

sayondeep, 2025-03-02:
@khuee The documentation has moved to https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/settings.html#other-frameworks-specific-settings