espressif/esp-matter #685: Error modifying the.zap file for zap light

KEEPHEADU, 2023-10-17:
**Describe the bug**
When I do not enable the Color Control cluster there will be compilation error in zap_light demo, I want to know how to crop the cluster in zap_light demo.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file): `Executing action: all (aliases: build)
Running ninja in directory /home/xh/matter_space/zap_light/build
Executing "ninja all"...
[1/101] Performing build step for 'chip_gn'ninja: no work to do.
[1/1] cd /home/xh/matter_space/zap_light/build/bootloader/esp-idf/esptool_py && /home/xh...et 0x8000 bootloader 0x0 /home/xh/matter_space/zap_light/build/bootloader/bootloader.binBootloader binary size 0x5030 bytes. 0x2fd0 bytes (37%) free.
[3/97] Generating ../../gen/app-zapgen/zapgen/app-templates/zap-generated/access.h, ../....g.h, ../../gen/app-zapgen/zapgen/app-templates/zap-generated/IMClusterCommandHandler.cpp🔧 Using temporary state directory: /tmp/zap0jrur9w9/zap.DcHogI
🤖 ZAP generation started: 
    🔍 input files: /home/xh/matter_space/test/light.zap
    🔍 output pattern: /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated
    🔍 using templates: /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/app-templates.json
    🔍 using zcl data: /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/zap-templates/zcl/zcl.json
    🔍 zap version: ver. 2023.7.20, featureLevel 97, commit: b57cef1fc261cedcbee418566a3d1e83b2d200d6 from 2023-07-20T18:42:00.000Z, mode: binary, exe: /home/xh/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/zap/zap-cli
🕐 Setup time: 561ms 
👉 using input file: /home/xh/matter_space/test/light.zap
👉 using output destination: /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated
🕐 File loading time: 65ms
🤖 Generating files:
    ✍  /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/CHIPClusters.h
    ✍  /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/IMClusterCommandHandler.cpp
    ✍  /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h
    ✍  /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/access.h
    ✍  /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/endpoint_config.h
🕐 Generation time: 410ms 
Searching for zcl file from /home/xh/matter_space/test/light.zap
Files generated in: /home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated
[4/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/esp_matter_command.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/af.h:23,
                 from /home/xh/esp-matter/components/esp_matter/esp_matter_command.cpp:22:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
<command-line>: note: this is the location of the previous definition
[5/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/esp_matter_core.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/af.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/CommandResponseHelper.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/general-diagnostics-server/general-diagnostics-server.h:23,
                 from /home/xh/esp-matter/components/esp_matter/esp_matter_core.cpp:31:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
<command-line>: note: this is the location of the previous definition
[7/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/esp_matter_client.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/binding-table.h:25,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/bindings/PendingNotificationMap.h:18,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/bindings/BindingManager.h:21,
                 from /home/xh/esp-matter/components/esp_matter/esp_matter_client.cpp:19:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
<command-line>: note: this is the location of the previous definition
/home/xh/esp-matter/components/esp_matter/esp_matter_client.cpp:192:13: warning: 'void esp_matter::cluster::send_command_failure_callback(void*, CHIP_ERROR)' defined but not used [-Wunused-function]
  192 | static void send_command_failure_callback(void *context, CHIP_ERROR error)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xh/esp-matter/components/esp_matter/esp_matter_client.cpp:187:13: warning: 'void esp_matter::cluster::send_command_success_callback(void*, const chip::app::DataModel::NullObjectType&)' defined but not used [-Wunused-function]
  187 | static void send_command_success_callback(void *context, const chip::app::DataModel::NullObjectType &data)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[8/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/esp_matter_attribute_utils.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/af.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.h:23,
                 from /home/xh/esp-matter/components/esp_matter/esp_matter_attribute_utils.cpp:23:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
[... 246 lines cut ...]
[39/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/xh/e...meip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server-callback.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/af.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server.h:30,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server-callback.cpp:24:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
<command-line>: note: this is the location of the previous definition
[40/97] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/descriptor/descriptor.cpp.objIn file included from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/config.h:29,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/af.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/util/attribute-storage.h:23,
                 from /home/xh/esp-matter/connectedhomeip/connectedhomeip/src/app/clusters/descriptor/descriptor.cpp:27:
/home/xh/matter_space/zap_light/build/gen/app-zapgen/zapgen/app-templates/zap-generated/gen_config.h:75: warning: "MATTER_SCENES_TABLE_SIZE" redefined
   75 | #define MATTER_SCENES_TABLE_SIZE 16
      | 
<command-line>: note: this is the location of the previous definition
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/xh/matter_space/zap_light/build/log/idf_py_stderr_output_270548 and /home/xh/matter_space/zap_light/build/log/idf_py_stdout_output_270548
`
- Host Machine OS:Linux
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

jonsmirl, 2023-10-17:
The esp-matter light example is much easier to use.
https://github.com/espressif/esp-matter/tree/main/examples/light

jadhavrohit924, 2023-10-18:
@KEEPHEADU Can you please add all the step that you have followed? Also, please attached ( do not paste ) the console logs for better analysis.

KEEPHEADU, 2023-10-18:
> The esp-matter light example is much easier to use. You don't need to use zap. https://github.com/espressif/esp-matter/tree/main/examples/light

This is a good suggestion, I tried to add cluster and attribute using code before, it was more organized than zap tool, I will try to add the cluster and attribute I need using code now, thanks.

KEEPHEADU, 2023-10-18:
> @KEEPHEADU Can you please add all the step that you have followed? Also, please attached ( do not paste ) the console logs for better analysis.

When I used the zap tool to disable the color control cluster on the zap file in the zap_light demo, it gave me a compilation error, but when I added the cluster in this demo, there was no problem.

jadhavrohit924, 2023-10-18:
To solve the actual issue you have to add `idf_build_set_property(CXX_COMPILE_OPTIONS "-DEMBER_AF_COLOR_CONTROL_CLUSTER_SERVER_ENDPOINT_COUNT=1" APPEND)` in [here](https://github.com/espressif/esp-matter/blob/main/examples/zap_light/CMakeLists.txt#L57)

KEEPHEADU, 2023-10-18:
> To solve the actual issue you have to add `idf_build_set_property(CXX_COMPILE_OPTIONS "-DEMBER_AF_COLOR_CONTROL_CLUSTER_SERVER_ENDPOINT_COUNT=1" APPEND)` in [here](https://github.com/espressif/esp-matter/blob/main/examples/zap_light/CMakeLists.txt#L57)

This instruction is valid. Can you tell me the meaning of this instruction? Thank you

KEEPHEADU, 2023-10-18:
> To solve the actual issue you have to add `idf_build_set_property(CXX_COMPILE_OPTIONS "-DEMBER_AF_COLOR_CONTROL_CLUSTER_SERVER_ENDPOINT_COUNT=1" APPEND)` in [here](https://github.com/espressif/esp-matter/blob/main/examples/zap_light/CMakeLists.txt#L57)

Now the project compiles correctly when I disable the color control cluster, and chip-tool can not read the color control cluster related information.

wqx6, 2023-10-18:
> This instruction is valid. Can you tell me the meaning of this instruction? Thank you

@KEEPHEADU The esp_matter component adds all of the sub-directories under `connectedhomeip/srs/app/clusters/` to its SRCS_DIRS. But the zap_light only need a few of those clusters sub-directories and these macro definitions will not be generated in the zap-generated files. The source files under those directories will be compiled for zap-light which will leads to compiling errors for those files. So we need to add the marcos to avoid the errors.

KEEPHEADU, 2023-10-19:
> @KEEPHEADU The esp_matter component adds all of the sub-directories under `connectedhomeip/srs/app/clusters/` to its SRCS_DIRS. But the zap_light only need a few of those clusters sub-directories [...]

I see. Thank you

KEEPHEADU, 2023-10-19:
Closing the problem

closed 2023-10-19 by KEEPHEADU