espressif/esp-matter #686: Zigbee Bridge

antonio-gabriele, 2023-10-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: 79dfcc7b9db37d80105e2bbb39a680a0670296d5
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP-S3
- Device Logs (Please attach the log file):
- Host Machine OS: ubuntu
- Commissioner app and versions if present:
- Commissioner's logs if present:

_Hi, as in the additional details, I flash zigbee rcp and zigbee bridge application and get this error loop on wifi_

**Any additional details**
I (1601) chip[DL]: WIFI_EVENT_STA_START
W (1601) wifi:Haven't to connect to a suitable AP now!
I (1611) chip[DL]: Done driving station state, nothing else to do...
I (1621) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1621) chip[DL]: Device already advertising, stop active advertisement and restart
I (1631) NimBLE: GAP procedure initiated: stop advertising.

I (1641) NimBLE: GAP procedure initiated: advertise;
I (1641) NimBLE: disc_mode=2
I (1651) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1661) NimBLE:

I (1661) app_main: Commissioning window opened
I (1761) main_task: Returned from app_main()
                                                                                                                                                                                                                                                                                                                                                                                            I (1801) esp_zboss: status: -1
I (1831) esp_zboss: Zigbee stack initialized
> I (31531) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31541) chip[DL]: Device already advertising, stop active advertisement and restart
I (31551) NimBLE: GAP procedure initiated: stop advertising.

I (31551) NimBLE: GAP procedure initiated: advertise;
I (31561) NimBLE: disc_mode=2
I (31561) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31581) NimBLE:

I (38401) wifi:I (38401) chip[DL]: flush txqWIFI_EVENT_STA_STOP

I (38401) wifi:I (38401) chip[DL]: stop sw txqStarting ESP WiFi layer

**E (38411) chip[DL]: esp_wifi_start() failed: ESP_ERR_WIFI_STOP_STATE**
I (38401) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037e8b2
0x4037e8b2: esp_cpu_wait_for_intr at /home/toor/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Oct 17 2023 16:10:53
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...

wqx6, 2023-10-18:
Not sure why the device failed to connect the Wi-Fi AP. Could you please upload the full log of the device?

antonio-gabriele, 2023-10-19:
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Oct 17 2023 16:10:53
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (77) boot:  1 nvs              WiFi data        01 02 00010000 0000e000
I (84) boot:  2 nvs_keys         NVS keys         01 04 0001e000 00001000
I (92) boot:  3 otadata          OTA data         01 00 0001f000 00002000
I (99) boot:  4 phy_init         RF data          01 01 00021000 00001000
I (107) boot:  5 ota_0            OTA app          00 10 00030000 001d0000
I (114) boot:  6 ota_1            OTA app          00 11 00200000 001d0000
I (122) boot:  7 fctry            WiFi data        01 02 003d0000 00006000
I (129) boot:  8 zb_storage       Unknown data     01 81 003d6000 00020000
I (137) boot:  9 zb_fct           Unknown data     01 81 003f6000 00000400
I (144) boot: End of partition table
I (149) esp_image: segment 0: paddr=00030020 vaddr=3c140020 size=3bea0h (245408) map
I (201) esp_image: segment 1: paddr=0006bec8 vaddr=3fc99d00 size=04150h ( 16720) load
I (205) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=130e34h (1248820) map
I (431) esp_image: segment 3: paddr=001a0e5c vaddr=3fc9de50 size=00b1ch (  2844) load
I (432) esp_image: segment 4: paddr=001a1980 vaddr=40374000 size=15cb8h ( 89272) load
I (466) boot: Loaded app from partition at offset 0x30000
I (466) boot: Disabling RNG early entropy source...
I (477) cpu_start: Multicore app
I (478) cpu_start: Pro cpu up.
I (478) cpu_start: Starting app cpu, entry point is 0x4037556c
I (0) cpu_start: App cpu up.
I (496) cpu_start: Pro cpu start user code
I (496) cpu_start: cpu freq: 160000000 Hz
I (496) cpu_start: Application information:
I (499) cpu_start: Project name:     zigbee_bridge
I (504) cpu_start: App version:      1.0
I (509) cpu_start: Compile time:     Oct 17 2023 16:10:31
I (515) cpu_start: ELF file SHA256:  ea6d0c424aecbecd...
I (521) cpu_start: ESP-IDF:          v5.1.1-dirty
I (526) cpu_start: Min chip rev:     v0.0
I (531) cpu_start: Max chip rev:     v0.99
I (536) cpu_start: Chip rev:         v0.2
I (541) heap_init: Initializing. RAM available for dynamic allocation:
I (548) heap_init: At 3FCB54A8 len 00034268 (208 KiB): DRAM
I (554) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (561) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (567) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (574) spi_flash: detected chip: generic
I (578) spi_flash: flash io: dio
I (583) sleep: Configure to isolate all GPIO pins in sleep state
I (589) sleep: Enable automatic switching of GPIO sleep configuration
I (596) coexist: coex firmware version: 80b0d89
I (601) coexist: coexist rom version e7ae62f
I (606) app_start: Starting scheduler on CPU0
I (611) app_start: Starting scheduler on CPU1
I (611) main_task: Started on CPU0
I (621) main_task: Calling app_main()
I (661) pp: pp rom version: e7ae62f
I (661) net80211: net80211 rom version: e7ae62f
I (671) wifi:wifi driver task: 3fcc2314, prio:23, stack:6656, core=0
I (671) wifi:wifi firmware version: ce9244d
I (671) wifi:wifi certification version: v7.0
I (671) wifi:config NVS flash: enabled
[... 1950 lines cut ...]
I (1476) chip[DIS]: CHIP minimal mDNS started advertising.
I (1486) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (1496) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 548556BDFD320131.
I (1506) chip[DIS]: mDNS service published: _matterc._udp
I (1506) chip[IN]: CASE Server enabling CASE session setups
I (1516) chip[SVR]: Joining Multicast groups
I (1516) chip[SVR]: Server Listening...
I (1526) esp_matter_core: Dynamic endpoint 0 added
I (1526) esp_matter_core: Dynamic endpoint 1 added
I (1536) chip[DL]: WIFI_EVENT_STA_START
W (1536) wifi:Haven't to connect to a suitable AP now!
I (1546) chip[DL]: Done driving station state, nothing else to do...
I (1556) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1556) chip[DL]: Device already advertising, stop active advertisement and restart
I (1566) NimBLE: GAP procedure initiated: stop advertising.

I (1576) NimBLE: GAP procedure initiated: advertise;
I (1576) NimBLE: disc_mode=2
I (1586) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1596) NimBLE:

I (1596) app_main: Commissioning window opened
                                                                                                                                                                                                                                            I (1756) main_task: Returned from app_main()
I (1796) esp_zboss: status: -1
> I (1826) esp_zboss: Zigbee stack initialized

wqx6, 2023-11-23:
Oh, There is an issue for the [PIN map](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge#11-hardware-connection), The PIN 4 of ESP32/ESP32-S3 should be connected to the TX PIN of ESP32-H2, and the PIN 5 of ESP32/ESP32-S3 should be connected to the RX PIN of ESP32-H2. Will update the doc ASAP.

antonio-gabriele, 2023-12-02:
Yes, change above resolved issue.

closed 2023-12-02 by antonio-gabriele