espressif/esp-matter #688: 关于 桥接设备 场景创建的问题

wnylei, 2023-10-18:
我现在想要实现把 一个非matter系统中的场景, 桥接到苹果的家庭APP中.  但在实现过程中,由于自身技术能力不够的问题,卡在了某个问题上. 希望可以得到大家的指点.

首先,我通过matter核心文档,了解到可以通过 Actions Cluster 中action_list这个属性 添加 list[Action Struct]数据  来实现此功能. 
然后,我在esp的组件中找到了下面这个创建  action_list 的函数. 

  `attribute_t *create_action_list(cluster_t *cluster, uint8_t *value, uint16_t length, uint16_t count);`

但是问题来了 ,这个函数中 uint8_t *value  这个入参,对应的结构体变量应该怎么创建和添加数据, 我在官方例程中没有找到关于 list[Action Struct]  对应的结构体.  
想要请问:
1,是需要自己定义  list[Action Struct] 这个结构体吗,如果是可否给一个示例?
2,如果不是,应该如何实现这个结构体 list[Action Struct] 变量中对应的属性数据 赋值和更新?

期待大佬的指点.......

wqx6, 2023-10-18:
Actions cluster 目前在connectedhomeip 这个upstream repo里还没有相应的支持，所以我们目前也只是保持这个cluster能在我们的data model 是可创建的状态，具体功能的实现可能需要等upstream的支持，之后upstream应该也会提供创建以及更新list[Action Struct] 的API

wnylei, 2023-10-18:
非常感谢! 另外还有几个问题,求教一下.
1. 是不是目前没有办法实现,把一个非matter系统中的 场景, 桥接给到苹果的家庭APP中?      
2. 像一些复杂的属性, 如 list[End pointList Struct] 这个属性, 它可以对桥接设备进行分组,是不是同样没有得到connectedhomeip支持?           
3.  还有就是cluster或attribute 有没有得到connectedhomeip支持或不支持 可以在哪里有文档可以看到吗?

wqx6, 2023-10-19:
> 1. 是不是目前没有办法实现,把一个非matter系统中的 场景, 桥接给到苹果的家庭APP中?

这个是可以的，Action Cluster 只是Bridge的可选功能，我们是可以实现基础的bridge的，就是将Matter命令转换成bridged设备可理解的命令发送给bridged设备

> 2. 像一些复杂的属性, 如 list[End pointList Struct] 这个属性, 它可以对桥接设备进行分组,是不是同样没有得到connectedhomeip支持?

每个桥接设备在bridge上都会抽象成一个endpoint，它们一般都会有一个Aggregator parent endpoint，拓扑结构大致如下:
```
root_node---------Aggregator1---------BridgedDevice1
           |                   |
           |                   ------ BridgedDevice2
           |
           -----Aggregator2----------BridgedDevice3
```
也可以实现对桥接设备进行分组

> 3. 还有就是cluster或attribute 有没有得到connectedhomeip支持或不支持 可以在哪里有文档可以看到吗?

 这个之能看最近的test plan文档，或者是src/app/clusters 里的代码了

wnylei, 2023-10-19:
**非常感谢!**

可能我表述的问题不够准确, 我期望达到的效果是, 首先,我有一个非matter的智能家居系统,在这个系统中,已经存在了多个设备和场景.  
当我通过苹果的家庭 APP 添加matter桥接网关时, 苹果的家庭APP可以根据桥接网关提供的桥接信息,  展示出这些桥接设备和场景的UI, 以及这些设备和场景的分组名称(房间名称),.   这样可以有助于用户的快速使用.  

1. 关于第一个问题,同步场景的问题, 我想达到的效果是在苹果的家庭APP中直接就能展示出对应的场景对象UI. 这个需求,目前的matter是可行的吗?
2. 关于第二个问题,桥接设备和场景分组的问题, 我想达到的效果是,当桥接设备和场景同步到苹果的家庭APP中时, 这些设备和场景是已经分好了房间的.   这样用户在苹果的家庭APP中就不用再对设备进行二次区分房间了.  我也尝试了按照你说的这个结构使用多个Aggregator 来对设备进行分组. 但并没有达到想要的效果,而且出现了一些奇怪的问题. 基于这个需求,目前的matter是可行的吗?

wnylei, 2023-10-20:
**非常感谢!**

closed 2023-10-20 by wnylei