espressif/esp-matter #689: Just pulled, light won't compile

jonsmirl, 2023-10-18:
What is this?
~~~
e/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_feature.cpp
/home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_feature.cpp: In function 'uint32_t esp_matter::cluster::on_off::feature::dead_front_behavior::get_id()':
/home/jonsmirl/aosp/esp-matter/components/esp_matter/esp_matter_feature.cpp:340:38: error: 'kDeadFrontBehavior' is not a member of 'chip::app::Clusters::OnOff::Feature'
  340 |     return (uint32_t)OnOff::Feature::kDeadFrontBehavior;
      |                                      ^~~~~~~~~~~~~~~~~~
[36/175] Building CXX object esp-idf/esp_matter/CMake...omeip/connect
~~~

jonsmirl, 2023-10-18:
What is an ICD Manager and why does a light bulb need one?

jonsmirl, 2023-10-18:
I've commended it out so that I can restart tests running on most up-to-date light code.

wqx6, 2023-10-19:
Could you please update the submodule? We have update the connectedhomeip submodule recently. BTW, ICDManager is designed for Intermittently Connected Device(ICD) which is related to power save.

ahmed8812, 2023-10-19:
@wqx6 

How to update the submodules safely?  I did try to update to update submodules (connectedhomeip) but something is not going right
Can you provide the command?

Thanks

wqx6, 2023-10-19:
@ahmed8812 You can use these two commands:
```
git submodule sync --recursive
git submodule update --init --recursive
```

jonsmirl, 2023-10-19:
This commit is not in the CHIP repo
~~~
jonsmirl@ares:~/aosp/esp-matter$ git submodule update --init --recursive
fatal: transport 'file' not allowed
fatal: Fetched in submodule path 'connectedhomeip/connectedhomeip', but it did not contain 699ff65e065a28e4b7ee3a2d84110444b1709040. Direct fetching of that commit failed.
fatal: 
~~~

dhrishi, 2023-10-26:
@jonsmirl I am able to compile the light example. The CI runs the compilations for examples, before the code is pushed to Github. So, ideally it should work fine. Please close the issue if you are able to get it to work

jonsmirl, 2023-10-26:
I fixed it now. I was looking in the wrong place. There was nothing wrong with esp-matter. The problem was that I forgot I put a log statement in the CHIP repo for the PACKET_BUFFER error. That change was preventing the CHIP tree from updating.  

Git generated an awful error message:
~~~
jonsmirl@ares:~/aosp/esp-matter$ git submodule update --init --recursive
fatal: transport 'file' not allowed
fatal: Fetched in submodule path 'connectedhomeip/connectedhomeip', but it did not contain 699ff65e065a28e4b7ee3a2d84110444b1709040. Direct fetching of that commit failed.
fatal:
~~~

closed 2023-10-26 by jonsmirl