espressif/esp-matter #690: How to use modeselect cluster

KEEPHEADU, 2023-10-19:
**Describe the question/query that you have**
How to use the supported_mode attribute in modeselect cluster, when I created the modeselect cluster using `mode_select::create`, In the source code we will use `attribute::create_supported_modes(cluster, NULL, 0, 0);` Create the supported_mode attribute, but it does not fill in the actual parameters, I want to know what the parameters should be filled in, I can not use the chip-tool tool to read the relevant information of the supported_mode attribute when I fill in an integer array.

**Additional context**

...

jadhavrohit924, 2023-10-19:
@KEEPHEADU We have documented the usage of mode select cluster for the `supported_modes` attribute. Please take a look at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#mode-select

KEEPHEADU, 2023-10-19:
> @KEEPHEADU We have documented the usage of mode select cluster for the `supported_modes` attribute. Please take a look at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#mode-select

thanks for your reply，I will close this issue

closed 2023-10-19 by KEEPHEADU