espressif/esp-matter #691: Using AWS OTA with ESP-Matter

anuarrozman, 2023-10-19:
Im planning on using AWS OTA features with Matter. My OTA function is ready and has been tested. Ive created my AWS OTA as component in ESP-Matter. Any resources or help i can get regarding this thing ?

shubhamdp, 2023-10-27:
We have a some [docs](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/production.html#over-the-air-ota-updates) which talks about [Matter OTA](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/ota.md). It also includes how to use encrypted OTA, you can take a look at that.

You may need to increment version number which matter basic information cluster reads. Take a look at [this](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-12-firmware-version-number) FAQ.

anuarrozman, 2023-10-30:
Thanks a bunch !! Ive done ESP AWS OTA demo and it works just fine. I've managed to make the AWS OTA as a component in my Matter project folder. When running with Matter, it has problem on connecting to my AWS endpoint which is not an issue when running AWS OTA without Matter.

anuarrozman, 2023-10-30:
From my reading, Matter CSA allows any method of OTA update right ? Not strictly bind to their OTA method.

shubhamdp, 2023-11-01:
> 13.5. Firmware
a. Nodes SHALL support OTA firmware updates, either using Matter-provided means (see Section 11.19, “Over-the-Air (OTA) Software Update”) or proprietary means. [CM58 for T59, T226]

This is from Matter Core Specification v1.2.

There has to be some means to OTA the device either Matter OTA or proprietary one.

dhrishi, 2023-11-01:
Closing as the question is answered

shubhamdp, 2023-11-01:
> Thanks a bunch !! Ive done ESP AWS OTA demo and it works just fine. I've managed to make the AWS OTA as a component in my Matter project folder. When running with Matter, it has problem on connecting to my AWS endpoint which is not an issue when running AWS OTA without Matter.

Can you check the free memory when using Matter. You can refer the below snippet for getting the values.
https://github.com/espressif/esp-matter/blob/9bada615639ebdb90966164d623c3c05ca16fac3/components/esp_matter_console/esp_matter_console_diagnostics.cpp#L27-L38

dhrishi, 2023-11-01:
> When running with Matter, it has problem on connecting to my AWS endpoint which is not an issue when running AWS OTA without Matter.

Sorry for missing this. I have reopened the issue

anuarrozman, 2023-11-02:
Hiii, Thanks for the responses. 

> This is from Matter Core Specification v1.2.
>
> There has to be some means to OTA the device either Matter OTA or proprietary one.

Yeap, Ive read this and think this could work. 

> Can you check the free memory when using Matter. You can refer the below snippet for getting the values. [...]

For this, Ive check the heap memory using `ESP_LOGI(TAG, "[APP] Free memory: %"PRIu32" bytes", esp_get_free_heap_size());` this one is from ESP-IDF. When running both Matter and AWS, free heap memory during establishing connection to AWS is around 40000 - 50000 bytes. Without Matter, establishing connection to AWS will left around 100000+ bytes of free heap memory. Can this be the cause ? Sometimes when esp try to reestablish the AWS connection to the endpoint, the free heap memory become 900 bytes only, this is status when running with Matter.

anuarrozman, 2023-11-02:
> Can you check the free memory when using Matter. You can refer the below snippet for getting the values. [...]

Thanks @shubhamdp , I currently looking at this.

shubhamdp, 2023-11-06:
> the free heap memory become 900 bytes only, this is status when running with Matter

This could be a problem. You may need to optimize the RAM usage. There's a [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html) from esp-idf to minimize the RAM usage. There's another [blog post](https://blog.espressif.com/esp32-memory-analysis-case-study-eacc75fe5431) which talks about the minimizing the memory. Please take a look at them and see if you can reduce the RAM usage.

Also, can you share the following details, which could help to try out this situation?
**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- AWS IoT Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

anuarrozman, 2023-11-17:
Hi @shubhamdp, Im so sorry for the late response. I tried to recreate the RAM issues as well as unable to reconnect to AWS endpoint after device completed the update but suddenly no RAM issue. RAM start at around 199Kb at the start of the firmware, once ble deinit (matter) and RAM value at the start of OTA function is around 85Kb to 90Kb which is enough. I cant find my previous log where the problem occured but I'll just provide with you guys the commit hash of my ESP-Matter and ESP-IDF.  I'll also provide the same RAM issue on my team environment. 

Here are my environment details:
ESP-Matter Commit Id: 198db1245da2d48aca02ce544099d03f70368a6b
ESP-IDF Commit Id: ea5e0ff298e6257b31d8e0c81435e6d3937f04c7
AWS IoT Commit Id:  d37fd63002b4fda99523e1ac4c9822fce485e76d
SoC (eg: ESP32 or ESP32-C3): ESP32S3

Here are one of my team environment details:
ESP-Matter Commit Id: cd325c6fa4203c45707ecb8b5351b7303347acb7
ESP-IDF Commit Id: ea5e0ff298e6257b31d8e0c81435e6d3937f04c7
AWS IoT Commit Id:  d37fd63002b4fda99523e1ac4c9822fce485e76d
SoC (eg: ESP32 or ESP32-C3): ESP32S3

The log i provide is one of my team, since I cant recreate the issue.
[idf_py_stdout_output_68121.txt](https://github.com/espressif/esp-matter/files/13388591/idf_py_stdout_output_68121.txt)

dhrishi, 2024-12-05:
@anuarrozman With the latest main branch, some RAM optimizations are done. Please check those out. Please close the issue if resolved.

closed 2025-03-18 by dhrishi