espressif/esp-matter #692: The sleepy_device demo failed and kept restarting

KEEPHEADU, 2023-10-20:
**Is your feature request related to a problem? Please describe.**
While I was compiling and running the sleepy-device demo, the chip kept restarting. Here's the log of that time.
The chip I used was esp32h2 and I updated to the latest esp-idf and esp-matter.

**Describe the solution you'd like**
Description of what you want to happen.

**Describe alternatives you've considered**
Description of any alternative solutions or features you've considered.

**Additional context**
[log.txt](https://github.com/espressif/esp-matter/files/13051817/log.txt)

KEEPHEADU, 2023-10-23:
Now when I use chip-tool for networking, the Bluetooth connection always drops and reconnects until chip-tool times out.

wqx6, 2023-10-23:
Thanks for your report, it is an issue about the external crystal. And we will fix this issue ASAP.

KEEPHEADU, 2023-10-25:
@wqx6 Hello, is this problem solved now?

wqx6, 2023-10-25:
@KEEPHEADU Yes, this problem is solved now.

KEEPHEADU, 2023-10-26:
@wqx6 Ok, I'll close the issue.

KEEPHEADU, 2023-10-27:
@wqx6 Hello, I updated the latest esp-idf and esp-matter versions, but the chip still keeps restarting when running the sleepy_device example. Below is the log.
esp-matter branch : main 8b28a13 Merge branch 'docs/prerequisites_and_WSL' into 'main'
esp-idf branc: remotes/origin/release/v5.1 8c6114d0d2 Merge branch 'feature/free_bt_memory_v5.1' into release/v5.1
[log.txt](https://github.com/espressif/esp-matter/files/13186995/log.txt)

wqx6, 2023-10-27:
Sorrt for that there is an issue on openthread port which has been fixed in https://github.com/espressif/esp-idf/commit/1dc2a98d1eef76ff5bba838bcf0c93e64a5caa1b on ESP-IDF master, but this commit is still not back-port to release v5.1. Could you cherry-pick the commit and try again?

KEEPHEADU, 2023-10-27:
> 遗憾的是，在 ESP-IDF 主站的乐[鑫/ESP-idf@1dc2a98](https://github.com/espressif/esp-idf/commit/1dc2a98d1eef76ff5bba838bcf0c93e64a5caa1b) 中修复了 openthread 端口上的问题，但此提交仍然没有向后移植到 v5.1 版本。你能挑选提交并重试吗？

Ok, I'll switch and try again.

KEEPHEADU, 2023-10-27:
@wqx6 Hello, I switched esp-idf to 1dc2a98 commit in master branch, but when I compile the esp-matter example, I get an error, the error message is the same for every example, below is the log.

esp-idf version: 1dc2a98d1e (HEAD -> master) HEAD@{0}: reset: moving to 1dc2a98d1eef76ff5bba838bcf0c93e64a5caa1b
esp-matter version: 8b28a13 (HEAD -> main, origin/main, origin/HEAD) HEAD@{0}: pull: Fast-forward

[log.txt](https://github.com/espressif/esp-matter/files/13187683/log.txt)

KEEPHEADU, 2023-10-27:
After switching branches, I reexecuted./install.sh and the esp-idf example compiled correctly, as did the esp-matter example./install.sh, but the esp-matter example did not compile correctly.

KEEPHEADU, 2023-10-30:
> @wqx6 Hello, I switched esp-idf to 1dc2a98 commit in master branch, but when I compile the esp-matter example, I get an error, the error message is the same for every example, below is the log. [...]

I solved this problem by changing the TestCert_PAA_NoVID_Cert variable to constexprs and it worked.

closed 2023-10-30 by KEEPHEADU