espressif/esp-matter #694: Not load the header that newly created the component using "ESP-IDF: Create new ESP-IDF component"!

mohankrishkumare, 2023-10-25:
Not load the header that newly created the component using "ESP-IDF: Create new ESP-IDF component"!
How can I fix this?

"/home/mohan/Desktop/myCode/light/main/app_main.cpp:26:10: fatal error: myTest.h: No such file or directory
   26 | #include "myTest.h"
      |          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed."

**Environment**
- ESP-Matter Commit Id: 79dfcc7b9db37d80105e2bbb39a680a0670296d5
- ESP-IDF Commit Id: 79dfcc7b9db37d80105e2bbb39a680a0670296d5
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file): [log.txt](https://github.com/espressif/esp-matter/files/13165286/log.txt)
- Host Machine OS: Ubuntu 22.04.3 LTS
- Commissioner app and versions if present: 
- Commissioner's logs if present:

shubhamdp, 2023-10-26:
@mohankrishkumare If you have a component (eg: `myTest`) and want to use that in other component (eg: `main`), then you have to specify the dependency in the CMakeLists.txt. Please take a look at [component requirement docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements) from esp-idf.

dhrishi, 2023-11-07:
@mohankrishkumare Please close the issue if your problem is resolved

closed 2023-11-10 by mohankrishkumare