espressif/esp-matter #702: When I carry the "./install.sh" ,it can't generate the tool successfully

tianrongqin, 2023-10-28:
When I execute in the esp material directory/ When install. sh, it will get stuck during the installation of the tool and automatically exit the terminal, causing me to not be able to generate the required chip tool and not be able to use it. How can I solve this situation? Previously, we followed the SDK operation on the official website. Could you please answer the following questions?

tianrongqin, 2023-10-28:
Now when I implement "./install.sh" in esp-matter,it reported an error"gcc:not found".I also implement "sudo apt-get gcc",but it does not have any effect.

tianrongqin, 2023-10-28:
Now the error:
Building host tools

Done. Made 7423 targets from 346 files in 688ms
ninja: Entering directory `/home/tianrq/esp-matter/connectedhomeip/connectedhomeip/out/host'
ninja: error: '../../third_party/nlunit-test/repo/src/nlunit-test.c', needed by 'obj/third_party/nlunit-test/repo/src/libnlunit-test.nlunit-test.c.o', missing and no known rule to make it

jonsmirl, 2023-10-28:
Follow these instructions:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup

tianrongqin, 2023-10-29:
This is what I did, but it never worked.

tianrongqin, 2023-10-29:
When I follow this operation, I get the following error:

Building host tools

Done. Made 7423 targets from 346 files in 553ms
ninja: Entering directory `/home/tianrq/esp-matter/connectedhomeip/connectedhomeip/out/host'
ninja: error: '../../third_party/nlunit-test/repo/src/nlunit-test.c', needed by 'obj/third_party/nlunit-test/repo/src/libnlunit-test.nlunit-test.c.o', missing and no known rule to make it

jonsmirl, 2023-10-29:
I have those files.
~~~
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlunit-test/repo/src$ ls
Makefile.am  nltest.h       nlunit-test-config.h.in
Makefile.in  nlunit-test.c  nlunit-test.h
~~~

Check out the Chinese version of https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup
If you are inside China you have to do special things because of the China firewall.
Espressif people will help with this on Monday when they are at work.

wqx6, 2023-10-30:
Could you check the files under the directory `connectedhomeip/connectedhomeip/third_party/nlunit-test/repo/src`? Looks like the nlunit-test submodule is not cloned. Could you use the following commands to update the submodule?
```
git submodule sync --recursive
git submodule update --recursive --init
```

dhrishi, 2023-11-22:
@tianrongqin Please close the issue if resolved

dhrishi, 2023-12-01:
Closing this now

closed 2023-12-01 by dhrishi