espressif/esp-matter #704: No API to override command callback

jonsmirl, 2023-10-29:
Attributes have:
~~~
esp_err_t set_override_callback(attribute_t *attribute, callback_t callback)
~~~
There is no parallel API for commands.

When implementing a bridge you have to be able to override the commands too.

jonsmirl, 2023-10-29:
Temporarily, I used some casting to get to the callback
~~~
typedef struct _command {
    uint32_t command_id;
    uint16_t flags;
    command::callback_t callback;
    struct _command *next;
} _command2_t;

cluster = cluster::get(ep, OnOff::Id);
_command2_t *command = (_command2_t *)command::get(cluster, OnOff::Commands::Toggle::Id, COMMAND_FLAG_ACCEPTED);
command->callback = lp_command_callback_toggle;
~~~

jonsmirl, 2023-11-14:
I created a PR to enable this behavior
https://github.com/espressif/esp-matter/pull/733

dhrishi, 2024-01-24:
PR #734 is merged.

closed 2024-01-24 by dhrishi