espressif/esp-matter #706: chip-tool distribution network failed

KEEPHEADU, 2023-10-30:
**Describe the bug**
When I do a provisioning, it always fails, log down here.

**Environment**
- ESP-Matter Commit Id: cc58a481d23bf390d80a887176761204733d8d3f
- ESP-IDF Commit Id: 1dc2a98d1eef76ff5bba838bcf0c93e64a5caa1b
- SoC ESP32-H2
- Device Logs (Please attach the log file) :
[log.txt](https://github.com/espressif/esp-matter/files/13200311/log.txt)

- Host Machine OS: Linux
- Commissioner app and versions if present: chip-tool
- Commissioner's logs if present:
[log_chip_tool.txt](https://github.com/espressif/esp-matter/files/13200312/log_chip_tool.txt)

**Any additional details**
...

wqx6, 2023-10-30:
The log shows that the ESP32-H2 received CASE sigma1 from `dd7c:1678:9067:0:e663:429a:b2ce:521f` which is a Global Unicast Address. And the ESP32-H2 has no route for this address. But normally the commissioner should use a unique local address as the source address to send the commands. Could you print the addresses list of the interface on the commissioner to look up whether there is an unique local address? If there is no unique local address, you could use the following commands to enable RA reception on your host.
```
sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra=2
sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra_rt_info_max_plen=128
sudo sysctl -w net.ipv6.conf.all.forwarding=1
```

KEEPHEADU, 2023-10-30:
> The log shows that the ESP32-H2 received CASE sigma1 from which is a Global Unicast Address. And the ESP32-H2 has no route for this address. But normally the commissioner should use a unique local address as the source address to send the commands. Could you print the addresses list of the interface on the commissioner to look up whether there is an unique local address? If there is no unique local address, you could use the following commands to enable RA reception on your host.`dd7c:1678:9067:0:e663:429a:b2ce:521f`
> 
> ```
> sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra=2
> sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra_rt_info_max_plen=128
> sudo sysctl -w net.ipv6.conf.all.forwarding=1
> ```

This is the result of executing ifconfig. My computer has two network cards. The ip of the border router is 192.168.8.219, but the ip of the commissioner is 192.168.10.198.That's the question?

```
$ ifconfig
ens33: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.198  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fe80::3dc:b34d:8877:7c40  prefixlen 64  scopeid 0x20<link>
        inet6 dd7c:1678:9067::9a1  prefixlen 128  scopeid 0x0<global>
        inet6 dd7c:1678:9067:0:ed0a:a846:cb28:580a  prefixlen 64  scopeid 0x0<global>
        inet6 dd7c:1678:9067:0:66d1:193f:7a9e:564  prefixlen 64  scopeid 0x0<global>
        ether 00:0c:29:e8:22:33  txqueuelen 1000  (以太网)
        RX packets 58895671  bytes 39961415560 (39.9 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 53209455  bytes 36628564112 (36.6 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (本地环回)
        RX packets 5368  bytes 1249027 (1.2 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5368  bytes 1249027 (1.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

KEEPHEADU, 2023-10-30:
@wqx6 It is this problem that when I switch the commissioner to the same network segment as the border router, the commissioning is normal.thanks

closed 2023-10-30 by KEEPHEADU