espressif/esp-matter #707: No way to set CHIP_CONFIG_MAX_GROUPS_PER_FABRIC

jonsmirl, 2023-10-30:
CHIP_CONFIG_MAX_GROUPS_PER_FABRIC defaults to 4 which is too low for my needs. 

How do I change it? This doesn't work in CMakeLists.txt
~~~
idf_build_set_property(C_COMPILE_OPTIONS "-DCHIP_CONFIG_MAX_GROUPS_PER_FABRIC=20" APPEND)
~~~

jonsmirl, 2023-10-30:
Actually I think I need CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC not CHIP_CONFIG_MAX_GROUPS_PER_FABRIC. Four is fine for CHIP_CONFIG_MAX_GROUPS_PER_FABRIC. I need to set CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC to 8 for my Keypad.

wqx6, 2023-11-07:
You could set this value by create a project_config header file and define this value in that file. And you should set the menuconfig option `CHIP_PROJECT_CONFIG` to project_config file's relative path to the `CMAKE_SOURCE_DIR`.

jonsmirl, 2023-11-11:
CHIP_PROJECT_CONFIG is in your project menuconfig
the use a path like "main/CHIPConfig.h"

closed 2023-11-11 by jonsmirl