espressif/esp-matter #708: Running CHIP on the TIMER task stack

jonsmirl, 2023-10-31:
In the generic switch example the button events are arriving on the TIMER task stack. Then routines like app_driver_button_release() proceed to call into CHIP on the TIMER task stack.  Shouldn't these button events be queued and then processed on the main stack? The purpose of the example is to show good coding practices. This change would illustrate how to drive the main task using events.

shripad621git, 2023-11-02:
@jonsmirl ,thanks for the suggested change. We are working on it. We will make the change available soon.

jonsmirl, 2023-11-09:
I have not fixed yet in my code, this is what it looks like when it fails.
~~~
I (42715) chip[ZCL]: SwitchServer: OnShortRelease
I (42725) chip[DIS]: Found an existing secure session to [2:AB0DE00000000037]!
I (42725) chip[-]: Unicast CB cluster 6 command 1
I (42725) chip[EM]: <<< [E:2930i S:19022 M:5288788] (S) Msg TX to 2:AB0DE00000000037 [4507] [UDP:[2600:4040:51BF:4000:D5ED:D58C:3B22:6B4%st1]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
I (42725) chip[DIS]: Found an existing secure session to [2:AB0DE00000000035]!
I (42725) chip[-]: Unicast CB cluster 6 command 1
I (42725) chip[EM]: <<< [E:2931i S:
***ERROR*** A stack overflow in task esp_timer has been detected.
~~~

jonsmirl, 2023-11-09:
You might just want to make a new IOT_Button class which is designed around a queue instead of callbacks. Another problem with the existing IOT_Button is that it does not allow for arrays of buttons. 
~~~
    button_config_t kp_cfg = {
        .type = BUTTON_TYPE_CUSTOM,
        .custom_button_config = {
            .active_level = 0,
            .button_custom_init = kp_init,
            .button_custom_get_key_value = kp_get_key_value,
            .button_custom_deinit = kp_deinit,
            .priv = this,
        },
~~~
kp_get_key_value() doesn't allow return of a multiple button ID (ie a keypad) so I have to use a global variable as a side channel to pass the keypad key number.

jonsmirl, 2024-05-28:
If you get into excessive stack use problems, the solution is to use FreeRTOS messages to unwind the stacks.

closed 2024-05-28 by jonsmirl