espressif/esp-matter #711: Unable to build light project with Espressif mDNS

law-ko, 2023-11-01:
**Describe the bug**
When the `Use the minimal mDNS implementation shipped in the CHIP library` is unchecked, the `light` project will not be able to build.
[image]

**Environment**
- ESP-Matter Commit Id: `9bada615639ebdb90966164d623c3c05ca16fac3`
- ESP-IDF Commit Id: `6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f`
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Host Machine OS:Linux
- Device Logs (Please attach the log file):
```
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp: In function 'CHIP_ERROR chip::Dnssd::OnResolveDone(ResolveContext*)':
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp:472:83: error: use of deleted function 'chip::Span<T>::Span(std::nullptr_t, size_t) [with T = chip::Inet::IPAddress; std::nullptr_t = std::nullptr_t; size_t = unsigned int]'
  472 |         ctx->mResolveCb(ctx->mCbContext, nullptr, Span<Inet::IPAddress>(nullptr, 0), error);
      |                                                                                   ^
In file included from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/BufferWriter.h:24,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/system/SystemPacketBuffer.h:32,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BleLayer.h:59,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/CHIPCore.h:35,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/Optional.h:28,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/dnssd/platform/Dnssd.h:34,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.h:18,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp:18:
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/Span.h:57:5: note: declared here
   57 |     Span(std::nullptr_t null, size_t size) = delete;
      |     ^~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp: In function 'void chip::Dnssd::MdnsQueryDone(intptr_t)':
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp:533:98: error: use of deleted function 'chip::Span<T>::Span(std::nullptr_t, size_t) [with T = chip::Inet::IPAddress; std::nullptr_t = std::nullptr_t; size_t = unsigned int]'
  533 |                 resolveCtx->mResolveCb(ctx->mCbContext, nullptr, Span<Inet::IPAddress>(nullptr, 0), CHIP_ERROR_INVALID_ARGUMENT);
      |                                                                                                  ^
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/Span.h:57:5: note: declared here
   57 |     Span(std::nullptr_t null, size_t size) = delete;
      |     ^~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp:555:106: error: use of deleted function 'chip::Span<T>::Span(std::nullptr_t, size_t) [with T = chip::Inet::IPAddress; std::nullptr_t = std::nullptr_t; size_t = unsigned int]'
  555 |                         resolveCtx->mResolveCb(ctx->mCbContext, nullptr, Span<Inet::IPAddress>(nullptr, 0), CHIP_ERROR_NO_MEMORY);
      |                                                                                                          ^
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/Span.h:57:5: note: declared here
   57 |     Span(std::nullptr_t null, size_t size) = delete;
      |     ^~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/WiFiDnssdImpl.cpp:568:98: error: use of deleted function 'chip::Span<T>::Span(std::nullptr_t, size_t) [with T = chip::Inet::IPAddress; std::nullptr_t = std::nullptr_t; size_t = unsigned int]'
  568 |                 resolveCtx->mResolveCb(ctx->mCbContext, nullptr, Span<Inet::IPAddress>(nullptr, 0), CHIP_ERROR_INCORRECT_STATE);
      |                                                                                                  ^
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/Span.h:57:5: note: declared here
   57 |     Span(std::nullptr_t null, size_t size) = delete;
      |     ^~~~
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
[182/375] c++ obj/third_party/connected...rc/platform/ESP32/ESP32.DnssdImpl.cpp.oninja: build stopped: subcommand failed.
[1030/1226] Building C object esp-idf/e...es/__idf_espressif__mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/espuser/esp-git/esp-matter/examples/light/build/log/idf_py_stderr_output_176459 and /home/espuser/esp-git/esp-matter/examples/light/build/log/idf_py_stdout_output_176459
```

If Espressif mDNS cannot be used with Matter v1.2, any way to add DNS TXT record?
https://github.com/espressif/esp-matter/issues/563

shubhamdp, 2023-11-01:
@law-ko Please take a look at https://github.com/espressif/esp-matter/issues/697#issuecomment-1782412736.

We are in the process of updating the submodule and this will be fixed with that.

shubhamdp, 2023-11-06:
@law-ko This is fixed in https://github.com/espressif/esp-matter/commit/12716c76e7c0ddf13bddddb5e91a714129094436

closed 2023-11-06 by dhrishi