espressif/esp-matter #712: Errors messages from multiple endpoint with scenes cluster

jonsmirl, 2023-11-01:
I have eight endpoints with a scenes cluster so I get this set of error messages seven times. I think you track the initialization in other clusters so that these messages don't get generated.
~~~
E (1669) chip[ZCL]: Duplicate attribute override registration failed
E (1669) chip[ZCL]: Duplicate attribute override registration failed
E (1669) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1669) chip[ZCL]: Duplicate attribute override registration failed
E (1669) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: ScenesServer::Instance().Init() error: 3
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1679) chip[ZCL]: Duplicate attribute override registration failed
E (1689) chip[ZCL]: Duplicate attribute override registration failed
~~~

shripad621git, 2023-11-03:
@jonsmirl the error you are encountering basically is a check to prevent the reinitialization of the `scenes server`. On the `MatterScenesPluginServerInitCallback`, when a call to `ScenesServer::Init()` is made, the check is made and above error i.e Error 3  which is is `CHIP_ERROR_INCORRECT_STATE` is thrown as a indication.The code above is handled in `connectedhomeip.` Please find the references in the below links:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/scenes-server/scenes-server.cpp#L919
 https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/scenes-server/scenes-server.cpp#L107

jonsmirl, 2023-11-03:
I know it is not a real error. I thought your were adding code in esp-matter to suppress this message.

shripad621git, 2023-11-17:
@jonsmirl ,please close the issue if it is addressed.

closed 2023-11-17 by jonsmirl