espressif/esp-matter #714: Can't build IPv6 only (IPv4 turned off)

jonsmirl, 2023-11-02:
~~~
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp: In member function 'void chip::DeviceLayer::ConnectivityManagerImpl::UpdateInternetConnectivityState()':
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:42: error: 'netif_ip4_addr' was not declared in this scope; did you mean 'netif_ip6_addr'?
 1002 |                 if (!ip4_addr_isany_val(*netif_ip4_addr(netif)) && !ip4_addr_isany_val(*netif_ip4_gw(netif)))
      |                                          ^~~~~~~~~~~~~~
      |                                          netif_ip6_addr
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:42: note: maximum limit of 1000 namespaces searched for 'netif_ip4_addr'
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:22: error: 'ip4_addr_isany_val' was not declared in this scope; did you mean 'ip_addr_isany_val'?
 1002 |                 if (!ip4_addr_isany_val(*netif_ip4_addr(netif)) && !ip4_addr_isany_val(*netif_ip4_gw(netif)))
      |                      ^~~~~~~~~~~~~~~~~~
      |                      ip_addr_isany_val
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:22: note: maximum limit of 1000 namespaces searched for 'ip4_addr_isany_val'
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:89: error: 'netif_ip4_gw' was not declared in this scope; did you mean 'netif_is_up'?
 1002 |                 if (!ip4_addr_isany_val(*netif_ip4_addr(netif)) && !ip4_addr_isany_val(*netif_ip4_gw(netif)))
      |                                                                                         ^~~~~~~~~~~~
      |                                                                                         netif_is_up
/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ConnectivityManagerImpl_WiFi.cpp:1002:89: note: maximum limit of 1000 namespaces searched for 'netif_ip4_gw'
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
[319/339] c++ obj/ABS_PATH/home/jonsmirl/aosp/esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/ESP32_custom.ESP32DeviceInfoProvider.cpp.oninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonsmirl/lowpan/build/log/idf_py_stderr_output_1145466 and /home/jonsmirl/lowpan/build/log/idf_py_stdout_output_1145466

~~~

jonsmirl, 2023-11-02:
I want to build IPv6 only so that I can see if the group issue below is dependent on dual stacks.

shripad621git, 2023-11-03:
Can you please specify the options you enabled or disabled to build the app. Also please specify the commit id and the esp-idf version used.

jonsmirl, 2023-11-03:
esp-idf 6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f

Yes DISABLE_IPV4 - Disable IPv4 functionality in the CHIP stack
No LWIP_IPV6 - Enable IPv6

shripad621git, 2023-11-06:
Please do  specify the esp-matter commit id you are working on.

jonsmirl, 2023-11-06:
I am building the light example.

shripad621git, 2024-07-29:
@jonsmirl , we will be updating the submodule to include the change in the above PR [#33085](https://github.com/project-chip/connectedhomeip/pull/33085) in some time. Till then, you can apply the changes of the above PR on connectedhomeip submodule to test out the change.

dhrishi, 2024-09-02:
Fixed with the connectedhomeip submodule update.

closed 2024-09-02 by dhrishi