espressif/esp-matter #715: FailSafe timer fails commissioning early, "Can't reach Google," network works anyway

emmby, 2023-11-02:
I have an ESP32 using Matter https://github.com/project-chip/connectedhomeip/commit/08b13661b65f130d3bc7f7234e74e7c64ae2b3f7 (July 19 2023). Whenever I try to commission a test device onto Google Home, the FailSafeTimer expires well before 60 seconds (from section 5.5 of [the 1.0 spec](https://csa-iot.org/wp-content/uploads/2022/11/22-27349-001_Matter-1.0-Core-Specification.pdf)) have elapsed. See below where it expires in less than 45 seconds. When this happens, Google Home reports "Can't reach Google" and aborts the commissioning after the "Generating Matter Credentials," "Testing Network Connectivity", and "Connecting device to Google Home" steps.

[Troubleshooting for that error says](https://developers.home.google.com/matter/troubleshooting#commissioning_fails_with_a_could_not_contact_google_error_message) to check that I've create the proper PID/VID on the developer console, which I have and was working previously.

One thing that seems suspicious is that the failsafe timer is armed for just 1 second and then expires, aborting the commissioning. `Received ArmFailSafe (1s)`

The funny thing is that even though commissioning has failed, the device can access the network just fine. chip::DeviceLayer::ConfigurationMgr().IsFullyProvisioned() reports true and sntp network requests go through. But FabricTable.FabricCount() still reports 0.

The Google Home error message is new, I was not seeing that behavior previously even for the same device codebase. I'm wondering whether I got my fabric into a weird state somehow but I'm also not sure how to reset it. Clearing NVS on the esp32 and re-installing the Google Home app didn't resolve the problem.

Why won't this device commission onto the fabric?

* ESP32 Pico on Adafruit Feather ESP32 v2 using Wifi
* Yacubane esp32-arduino-matter v1.0.0-beta.6
* Matter 08b13661b65f130d3bc7f7234e74e7c64ae2b3f7
* esp-matter 603296a
* esp-idf v4.4.5
* arduino-esp32 2.0.11
* Android phone, Nest Home, Nest Wifi

[image]

emmby, 2023-11-02:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Wakeup was not caused by deep sleep: 0
Found PCT2075 temperature chip
State: entered main_region_HardwareInitialized
D (1251) esp_matter_core: Set val called with string with size 0
D (1252) esp_matter_core: Set val called with string with size 0
D (1255) esp_matter_core: Set val called with string with size 0
D (1261) esp_matter_core: Set val called with string with size 0
D (1267) esp_matter_core: Set val called with string with size 0
D (1273) esp_matter_core: Set val called with string with size 0
D (1279) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x6
D (1287) esp_matter_core: Set val called with string with size 0
D (1293) esp_matter_core: Set val called with string with size 0
D (1299) esp_matter_core: Set val called with string with size 0
D (1304) esp_matter_core: Set val called with string with size 0
D (1310) esp_matter_core: Set val called with string with size 0
D (1316) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x5
D (1325) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x5
D (1335) esp_matter_core: Set val called with string with size 0
D (1340) esp_matter_core: Set val called with string with size 0
D (1345) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x31, attribute_id-0x4
D (1354) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x31, attribute_id-0x4
D (1364) esp_matter_core: Set val called with string with size 0
D (1369) esp_matter_core: Set val called with string with size 0
D (1374) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x33, attribute_id-0x1
D (1383) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x33, attribute_id-0x1
D (1394) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x0
D (1401) esp_matter_core: Set val called with string with size 0
D (1407) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x0
D (1416) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x3
D (1425) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x3
D (1435) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x1
D (1442) esp_matter_core: Set val called with string with size 0
D (1448) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x1
D (1457) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x4
D (1466) esp_matter_core: Set val called with string with size 0
D (1472) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x4
D (1481) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3f, attribute_id-0x0
D (1489) esp_matter_core: Set val called with string with size 0
D (1495) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3f, attribute_id-0x0
D (1504) esp_matter_core: Set val called with string with size 0
D (1510) esp_matter_core: Set val called with string with size 0
D (1517) esp_matter_core: Set val called with string with size 0
D (1521) esp_matter_core: Set val called with string with size 0
D (1527) esp_matter_core: Set val called with string with size 0
D (1533) esp_matter_core: Set val called with string with size 0
D (1548) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x2a, attribute_id-0x0
D (1548) esp_matter_core: Set val called with string with size 0
D (1553) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x2a, attribute_id-0x0
I (1568) chip[DL]: NVS set: chip-counters/reboot-count = 1 (0x1)
I (1571) chip[DL]: NVS set: chip-counters/total-hours = 0 (0x0)
I (1575) chip[DL]: NVS set: chip-config/unique-id = "916F2021C37FA817"
I (1580) chip[DL]: Changing ESP WiFi mode: AP -> STA+AP
I (1585) chip[DL]: Changing ESP WiFi mode: STA+AP -> STA
I (1591) chip[DL]: Real time clock set to 946684800 (0099/11/31 16:00:00 UTC)
E (1596) esp_matter_providers: custom_dac_provider cannot be NULL
D (3682) chip[DL]: CHIP event task running
D (4007) CHIP[DL]: BLE Host Task Started
I (4033) CHIP[DL]: BLE host-controller synced
D (4539) CHIP[DL]: NimBLE start advertising...
I (4540) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (4540) NimBLE: GAP procedure initiated: advertise; 
[... 1918 lines cut ...]
I (46665) chip[DIS]: mDNS service published: _matterc._udp
D (46667) chip[DMG]: Endpoint 0, Cluster 0x0000_003E update version to 5b3c719e
D (46669) chip[DMG]: Endpoint 0, Cluster 0x0000_003E update version to 5b3c719f
D (46681) chip[DL]: Using hash:accaec5f6ac0c34 for nvs key:f/1/s/000000005356B4C2
D (46691) chip[DL]: Using hash:00f9871538fa9e7 for nvs key:g/s/BQpIEH30QHsT48nD/AY49A==
D (46778) chip[DL]: Using hash:accaec5f6ac0c34 for nvs key:f/1/s/000000005356B4C2
I (47298) chip[TS]: Pending Last Known Good Time: 2023-11-01T23:18:28
I (47299) chip[TS]: Previous Last Known Good Time: 2023-07-24T23:38:47
I (47300) chip[TS]: Reverted Last Known Good Time to previous value
[... cut ...]
D (47493) chip[DMG]: Endpoint 0, Cluster 0x0000_0030 update version to 72c86206
Commissioning failed, fail safe timer expired
E (47511) chip[DL]: Long dispatch time: 1688 ms, for event type 32783
I (47512) chip[FS]: Fail-safe cleanly disarmed
E (47515) chip[DL]: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E 3840
E (47521) chip[DL]: Configure Adv Data failed: Error ESP32:0x05B0001E
E (47528) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x05B0001E
Commissioning window opened
Fabric will be removed
Fabric removed successfully
Fabric will be removed```

shubhamdp, 2023-11-03:
@emmby Do you have a Nest Hub paired with your Google Home app, as that is the requirement. https://support.google.com/googlenest/answer/13127223?hl=en

shubhamdp, 2023-11-03:
@emmby Can you provide the version of your google home app, nest hub

emmby, 2023-11-03:
I have many Nest Home Hubs, Nest speakers, and Nest Wifi border routers on
my network. Is there a way to tell which one(s) you care about?

Google Home app is 3.7.1.4 on Android

shubhamdp, 2023-11-06:
I care about the version of Home Hub that is being used for adding this device

> We recommend that the account used for development only have a single Matter-enabled hub connected for testing. Multiple Matter-enabled hubs connected at the same time or connected to the same structure during testing could result in unpredictable behavior.

And since you have many there could be a problem, https://developers.home.google.com/matter/get-started#disclaimers

shubhamdp, 2023-12-07:
@emmby Is there any update on this?

emmby, 2023-12-07:
Closing as not reproducible for now

closed 2023-12-07 by emmby