espressif/esp-matter #717: sleepy end device crash loops after failed connection attempt

flashspys, 2023-11-02:
**Describe the bug**
I was able to run the sleepy end device example on a ESP32-H2. It works so far and my iPad can start the commissioning via the Home app. During the setup, the ESP32 crashes and crash loops from this point on. 

**Environment**
- ESP-Matter Commit Id: 5bbad2031440c8f45b2a655b002c5467286faa22
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32-H2
- Device Logs (Please attach the log file): [Terminal Saved Output.txt](https://github.com/espressif/esp-matter/files/13243842/Terminal.Saved.Output.txt)
- Host Machine OS: MacOS
- Commissioner app and versions if present: Home app (iOS 17.1)
- Commissioner's logs if present: -

wqx6, 2023-11-03:
Could you please use IDF on commit 6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f? The sleep support was not added on IDF v5.1.1 (e088c3766ba440e72268b458a68f27b6e7d63986).

dhrishi, 2023-11-06:
@flashspys Please give it a try. It should ideally work. Please close the issue if it does

dhrishi, 2023-12-08:
@flashspys Closing this. Please re-open in case of any issues

closed 2023-12-08 by dhrishi