espressif/esp-matter #720: Displays uncertified products on ios home

maenkai, 2023-11-07:
**Describe the bug**
My product has passed the matter certification, and I have purchased the official DAC certificate，
BUT Displays uncertified products on ios home

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
1.DCL
[image]

2.DAC
[dac.zip](https://github.com/espressif/esp-matter/files/13278125/dac.zip)

3.CMD
```
e.g:

    $matter_dir/tools/mfg_tool/mfg_tool.py -cn $meizu_pid_name\
    --vendor-name "$meizu_vid_name" \
    --product-name "$meizu_pid_name" \
    --hw-ver-str "$meizu_hd_version" \
    --mfg-date $meizu_date \
    -v $meizu_vid \
    -p $meizu_pid --pai \
    -c  $mfg_tool_pai_cert_pem \
    -cd $mfg_tool_dac_cd_der \
    --dac-key $mfg_tool_dac_key_pem \
    --dac-cert $mfg_tool_dac_cert_pem \
    --passcode $meizu_passcode \
    --discovery-mode $discover_mode
```

4.output Nvs
[partition.csv](https://github.com/espressif/esp-matter/files/13278154/partition.csv)

dhrishi, 2023-11-07:
@maenkai Are you using the CD file that you received from CSA after successful certification?

maenkai, 2023-11-08:
> @maenkai Are you using the CD file that you received from CSA after successful certification?

sure , I using the cd
[image]

shubhamdp, 2023-11-08:
@maenkai Does commissioning works irrespective of the warning pop up in iOS, or with chip-tool (assuming that you provide the `--paa-trust-store-path`)?

Can you please share the sdkconfig for this firmware?

maenkai, 2023-11-08:
> @maenkai Does commissioning works irrespective of the warning pop up in iOS, or with chip-tool (assuming that you provide the )?`--paa-trust-store-path`
> 
> Can you please share the sdkconfig for this firmware?

No problem, I can pair the device normally
This is my c
[sdkconfig.zip](https://github.com/espressif/esp-matter/files/13294236/sdkconfig.zip)
onfigure

dhrishi, 2023-11-22:
@maenkai Are you still seeing this issue, even after successfully updating the DCL model and also having the production CD in the device firmware?

dhrishi, 2023-11-30:
@maenkai Please close this issue if resolved

maenkai, 2023-12-04:
Log in. Enter the DCL Product Entries page in the Certification Tool. This problem can be solved

closed 2023-12-04 by maenkai