espressif/esp-matter #721: 'bool is_group' in command_handle

jonsmirl, 2023-11-08:
I don't think is_group should exist. The reason is that it is allowed to have a binding list which consist of a mix of groups and unicasts.  Code in esp-matter is preventing this from working when CHIP allows it.

In this case, you want the one that isn't NULL. The spec requires one or the other to be NULL. Just copy both of them?
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L790
~~~
        if (cmd->is_group) {
            this->group_id = cmd->group_id;
        } else {
            this->endpoint_id = cmd->endpoint_id;
        }
~~~
​Then in this case, you should just remove the check for is_group.
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_client.cpp#L119
~~~
    if (binding.type == EMBER_UNICAST_BINDING && !cmd_handle->is_group && peer_device) {
        if (client_command_callback) {
            cmd_handle->endpoint_id = binding.remote;
            client_command_callback(peer_device, cmd_handle, command_callback_priv_data);
        }
    } else if (binding.type == EMBER_MULTICAST_BINDING && cmd_handle->is_group && !peer_device) {
        if (client_group_command_callback) {
            cmd_handle->group_id = binding.groupId;
            client_group_command_callback(binding.fabricIndex, cmd_handle, command_callback_priv_data);
        }
    }
~~~
like this? Is esp-matter looping through all of the bindings? It is a list not a single item.  GROUP 12, NODE 3 EP 1, GROUP 2, NODE 7 EP 4 is an allowable binding list.
~~~
    if (binding.type == EMBER_UNICAST_BINDING && peer_device) {
        if (client_command_callback) {
            cmd_handle->endpoint_id = binding.remote;
            client_command_callback(peer_device, cmd_handle, command_callback_priv_data);
        }
    } else if (binding.type == EMBER_MULTICAST_BINDING) {
        if (client_group_command_callback) {
            cmd_handle->group_id = binding.groupId;
            client_group_command_callback(binding.fabricIndex, cmd_handle, command_callback_priv_data);
        }
    }
~~~

jonsmirl, 2023-11-08:
Why do I want this? Because multicast is unreliable and I am definitely losing group packets. So lets says I have lights A, B, C. I make a group G of {A, B, C}.  But then I bind my light switch to G, A, B, C.  When the switch is pressed the group command is sent first. But then it is followed up by individual unicasts to recover from dropped group messages.  I am not using commands which can't be sent twice like this. For example I only use on() or off(), never toggle().

jonsmirl, 2023-11-09:
This is going to take some rework to fix in esp-matter. This is likely the core of the problem...
~~~
typedef struct command_handle {
    union {
        uint16_t endpoint_id;
        uint16_t group_id;
    };
    uint32_t cluster_id;
    uint32_t command_id;
~~~
group_id and endpoint_id should not be unioned. They should be two separate variables. Then if group_id is set, it is a group command and if endpoint_id is set, it is a unicast command. Once they are split, then is_group is no longer needed.

Once they are split, when the binding callback happens you can set group_id and endpoint_id as needed for the binding. Then a binding which mixes groups and unicast will work.

chshu, 2023-11-13:
@jonsmirl The commit https://github.com/espressif/esp-matter/commit/12e73847aac2d80506295d12550bf9398ca2df62 should fix the issue, please check and close the issue.

jonsmirl, 2023-11-13:
In my five minutes of testing it appears to work. I will open new issue if I uncover deeper issues.

closed 2023-11-13 by jonsmirl