espressif/esp-matter #723: Random failed connection to the border router. (Role detached -> leader)

vvrec, 2023-11-11:
Random failed connection to HomePod Mini thread network.

- Two different official ESP32-H2 modules, one ESP32-H2 official  dev board.
- ESP-Matter v1.2
- ESP-IDF 5.2.0
- SoC  ESP32-H2 
- Device Logs :  [log.txt](https://github.com/espressif/esp-matter/files/13323782/log.txt)
- Host Machine OS: MacOS.
- Border router : Hompod mini.
- Commissioner app and versions if present: Apple iPhone Home app.
- sdkconfig - default for H2, partitions.csv - default or partitions_thread.csv 

Any example or own version of the code for h2 is successfully added to homekit, during several reloads of the microcontroller the role becomes the leader and does not change anymore. It happens with one connected device, two or three. Randomly on all devices. Accordingly, the accessory does not correspond in the Home app.

vvrec, 2023-11-12:
The problem was completely solved by adding a delay of at least 200 milliseconds before esp_matter::start.

Could this mean that reading from flash information critical for the thread (preliminary settings) is too slow? And how to solve it more elegantly than a delay?

vvrec, 2023-11-12:
The problem remains only when the microcontroller is rebooted with a soft reset.

VaishaliAvhale, 2023-11-23:
I have tried the same thing on my end using iPadOS stable version 17.1, and it is working fine. I'm in the process of testing this on iPadOS version 17.2, which you are using. I will update you once it is done. In the meantime, please update your HomePod to the latest version, 17.1, and try again. If you still face issues, please upload its detailed log files here.

vvrec, 2023-11-23:
> I have tried the same thing on my end using iPadOS stable version 17.1, and it is working fine. I'm in the process of testing this on iPadOS version 17.2, which you are using. I will update you once [...]

I don't think the problem is with the app for iOs... Perhaps you meant AudioOS HomePod Mini?
Anyway, I was able to work normally in the thread only with eps32-c6. Adding the lines from the generic switch example to the code:

```
nvs_handle_t handle;
nvs_open_from_partition(CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL, "chip-factory", NVS_READWRITE, &handle);
nvs_commit(handle);
nvs_close(handle);
```

It didn't change anything for H2. The device is connected to the network thread only after a few seconds without power, or a long press hard reset. With a normal reset or a soft reset, it becomes the sole leader and the impression is that the device does not read information about the thread network to which it has already joined.

vvrec, 2023-11-26:
I found the source of the problem, but no solution. The device cannot join the network thread after a hard reset if it is connected to the computer using the built-in USB d-/d+. I could not find sdkconfig option to solve this problem. However, when the device is not connected to USB, everything works as it should.

wqx6, 2023-11-27:
Could you please provide the commit id of esp-matter and esp-idf?

vvrec, 2023-11-27:
OK. Will do it again later today on a dev board on an unmodified example with the latest commits. And I will send a log

vvrec, 2023-11-27:
> Could you please provide the commit id of esp-matter and esp-idf?

IDF:```

commit 52bca70b1a73248c9a66095d2cfac8c6ebb1f263 (HEAD -> master)
Merge: e3eb2ea9cc 4e51c6b049
Author: Sudeep Mohanty <sudeep.mohanty@espressif.com>
Date:   Wed Aug 23 09:45:47 2023 +0800
```
Matter:```

commit d17b310177bbee3cf88197d54077c3458413085f (HEAD -> main)
Merge: 6680844 2322a30
Author: Hrishikesh Dhayagude <hrishi@espressif.com>
Date:   Fri Nov 17 18:20:14 2023 +0800
```

vvrec, 2023-11-27:
> I tried same thing with same specification on esp-matter latest commit and it is working properly at my end. Are you doing any change in code?

OK. Will do it again later today on a dev board on an unmodified example with the latest commits. And I will send a log

VaishaliAvhale, 2023-11-29:
For Matter projects development with this SDK, it is recommended to utilize ESP-IDF [v5.1 commit 6b1f40b9b](https://github.com/espressif/esp-idf/tree/6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f).
For more understanding check the given link: https://github.com/espressif/esp-matter

dhrishi, 2023-12-08:
@vvrec Do you have any update to share? Please close the issue if resolved

vvrec, 2023-12-08:
> @vvrec Do you have any update to share? Please close the issue if resolved

After the update, joining to thread started to work stably. But I still catch the "solo leader" with one condition that still needs to be tested. In my design, the device is powered by a LiPo backup. my unit is powered by usb + battery backup with a charge controller, then a 5V boost converter and then 3.3 volt step-down converter. The problem is that without using a boost to maintain 5 volts from the battery - the voltage can drop smoothly and when reset happens because brownout - the app restart and runs fully, but doesn't connect to the thread again until a forced reboot. If I turn off brownout reset in sdkconfig, the connection is successful. These conditions are difficult to reproduce when maintaining 5 volts, but the fact itself is there and sooner or later it may manifest itself.

wqx6, 2024-02-27:
I still cannot reproduce the issue. Could you please try to capture the Thread packets with the [instruction](https://openthread.io/guides/pyspinel/sniffer)? You can use an ESP32-H2 as the [RCP](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp#thread-sniffer)?

dhrishi, 2024-07-26:
I am closing this issue now. Please re-open if you see this again

closed 2024-07-26 by dhrishi