espressif/esp-matter #725: Fan Control Cluster::PercentSetting attribute error occur issue

YOGYUI, 2023-11-13:
**Describe the bug**
After commit id **d162b1357ce2d10a7971c434105633da0c4f3f11**, following error occurred when initializing Fan Control cluster.
```
E (6962) esp_matter_core: Cannot set bounds because of val type mismatch: expected: 136, min: 8, max: 8
```

With a little code inspection, I found the reason
```c
attribute_t *create_percent_setting(cluster_t *cluster, nullable<uint8_t> value, uint8_t min, uint8_t max)
{
    attribute_t *attribute =
	esp_matter::attribute::create(cluster, FanControl::Attributes::PercentSetting::Id,
                                         ATTRIBUTE_FLAG_NULLABLE | ATTRIBUTE_FLAG_WRITABLE,
                                         esp_matter_nullable_uint8(value));
    if (!attribute) {
        ESP_LOGE(TAG, "Could not create attribute");
        return NULL;
    }
    esp_matter::attribute::add_bounds(attribute, esp_matter_uint8(min), esp_matter_uint8(max));
    return attribute;
}
```
At **add_bounds** function in esp_matter_core.cpp, value types between attribute instance and function arguments should be matched. >> nullable_uint8 (type 136) mismatch with uint8 (type 8)

After some modification like below, error log is no longer appear.
- esp_matter_attribute.cpp
    ```c
    // attribute_t *create_percent_setting(cluster_t *cluster, nullable<uint8_t> value, uint8_t min, uint8_t max)
    attribute_t *create_percent_setting(cluster_t *cluster, uint8_t value, uint8_t min, uint8_t max)
    {
        attribute_t *attribute =
        esp_matter::attribute::create(cluster, FanControl::Attributes::PercentSetting::Id,
                                          ATTRIBUTE_FLAG_NULLABLE | ATTRIBUTE_FLAG_WRITABLE,
                                          //esp_matter_nullable_uint8(value));
                                          esp_matter_uint8(value));
  ```
-  esp_matter_attribute.h
    ```c
    namespace fan_control {
    namespace attribute {
    attribute_t *create_fan_mode(cluster_t *cluster, uint8_t value, uint8_t min, uint8_t max);
    attribute_t *create_fan_mode_sequence(cluster_t *cluster, const uint8_t value, uint8_t min, uint8_t max);
    // attribute_t *create_percent_setting(cluster_t *cluster, nullable<uint8_t> value, uint8_t min, uint8_t max);
    attribute_t *create_percent_setting(cluster_t *cluster, uint8_t value, uint8_t min, uint8_t max);
    attribute_t *create_percent_current(cluster_t *cluster, uint8_t value, uint8_t min, uint8_t max);
    ```
- esp_matter_cluster.h
    ```c
    namespace fan_control {
    typedef struct config {
        uint16_t cluster_revision;
        uint8_t fan_mode;
        uint8_t fan_mode_sequence;
        // nullable<uint8_t> percent_setting;
        uint8_t percent_setting;
        uint8_t percent_current;
        config() : cluster_revision(4), fan_mode(0), fan_mode_sequence(2), percent_setting(0), percent_current(0) {}
} config_t;
    ```

**Environment**
- ESP-Matter Commit Id: a7c1d960e9ad7ddbdb39e30a3f2d89ce622ef398
- ESP-IDF Commit Id: 6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Host Machine OS: Linux Ubuntu 22.04.3 LTS
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**

PSONALl, 2023-11-15:
@YOGYUI the percent setting is a nullable attribute, so the min and max bounds should be set to nullable instead,
I will add this fix soon!!

YOGYUI, 2023-11-15:
@PSONALl  I understood what you're saying. So, can i expect that <esp_matter_attribute.cpp> will be modified like below?
```c
attribute_t *create_percent_setting(cluster_t *cluster, nullable<uint8_t> value, uint8_t min, uint8_t max)
{
    attribute_t *attribute =
	esp_matter::attribute::create(cluster, FanControl::Attributes::PercentSetting::Id,
                                         ATTRIBUTE_FLAG_NULLABLE | ATTRIBUTE_FLAG_WRITABLE,
                                         esp_matter_nullable_uint8(value));
    if (!attribute) {
        ESP_LOGE(TAG, "Could not create attribute");
        return NULL;
    }
    // esp_matter::attribute::add_bounds(attribute, esp_matter_uint8(min), esp_matter_uint8(max));
    esp_matter::attribute::add_bounds(attribute, esp_matter_nullable_uint8(min), esp_matter_nullable_uint8(max));
    return attribute;
}
```

PSONALl, 2023-11-15:
yes @YOGYUI 

```
attribute_t *create_percent_setting(cluster_t *cluster, nullable<uint8_t> value, nullable<uint8_t> min, nullable<uint8_t> max)
{
    attribute_t *attribute =
	esp_matter::attribute::create(cluster, FanControl::Attributes::PercentSetting::Id,
                                         ATTRIBUTE_FLAG_NULLABLE | ATTRIBUTE_FLAG_WRITABLE,
                                         esp_matter_nullable_uint8(value));
    if (!attribute) {
        ESP_LOGE(TAG, "Could not create attribute");
        return NULL;
    }
    esp_matter::attribute::add_bounds(attribute, esp_matter_nullable_uint8(min), esp_matter_nullable_uint8(max));
    return attribute;
}
```

YOGYUI, 2023-11-16:
@PSONALl Thanks, I'll wait for next commit. :)

closed 2023-11-16 by YOGYUI