espressif/esp-matter #731: Matter over Thread Generic Switch Example Issue

bilalmalik76, 2023-11-14:
I am using Matter over Thread with  Generic Switch Example But I am getting this error. I have also flashed mfg_binaries but still getting the same error.
[image]

bilalmalik76, 2023-11-14:
Can anyone help me with this?

dhrishi, 2023-11-14:
@bilalmalik76 Can you please share all the details that are required when you choose "Raise a new issue". Also, a complete set of steps that you've followed for the mfg_tool and flashing the binaries would help as well

bilalmalik76, 2023-11-14:
@dhrishi I have followed these steps: 

1. set the target to esp32-c6: idf.py set-target esp32c6
2. Build the project with idf.py build
3.  Falsh the code to esp32c6 with idf.py flash -p /dev/ttyUSB0
4.  Flashed the mfg_binaries with: esptool.py -p dev/ttyUSB0 write_flash 0x10000 mfg_binaries/20202020_3841.bin
5.  At the end: idf.py monitor

dhrishi, 2023-11-14:
@bilalmalik76 Please provide the other details too:

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

bilalmalik76, 2023-11-14:
@dhrishi 
Environment

ESP-Matter Commit Id: 01bebca56b42a35b5ad049f59292df7d4c08ea6b
ESP-IDF Commit Id: cbce221e88d52665523093b2b6dd0ebe3f1243f1
SoC (eg: ESP32 or ESP32-C3): tried with ESP32-C6 and ESP32-H2, getting same error on both
Device Logs (Please attach the log file): Attached
[logs.txt](https://github.com/espressif/esp-matter/files/13352127/logs.txt)

Host Machine OS: Ubuntu 22
Commissioner app and versions if present: Apple Home App ioS 17.1.1
Commissioner's logs if present:

bilalmalik76, 2023-11-18:
@sayondeep I have not changed anything and using sdkconfig.defaults and sdkconfig.defaults.esp32c6 already available in example.

bilalmalik76, 2023-11-22:
@sayondeep any update on this?

VaishaliAvhale, 2023-11-22:
@bilalmalik76 following the instructions [here](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch). I was able to commission the generic_switch example on esp32c6 successfully.
I was able to reproduce the error when mfg_binaries was not flashed. Can you please try again in the following sequence:
1)idf.py -p <port> erase-flash
2)idf.py -p <port> flash
3)esptool.py -p [port-name] write_flash 0x10000 mfg_binaries/20202020_3841.bin

bilalmalik76, 2023-11-22:
@VaishaliAvhale I followed same steps earlier, let me try again and come back to you.

sayondeep, 2023-12-04:
@bilalmalik76 If the issue is resolved can you please close the issue.

closed 2023-12-08 by dhrishi