espressif/esp-matter #732: Matter ESP32-C6 Generic Switch build issue

lboue, 2023-11-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: 01bebca56b42a35b5ad049f59292df7d4c08ea6b
- ESP-IDF Commit Id: 66992aca7ae26afaede9933dab2b46e1bf326ba6
- SoC: ESP32-C6
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04.2 LTS (GNU/Linux 6.2.0-36-generic x86_64
- Commissioner app and versions if present: N/A. It doesn't build.
- Commissioner's logs if present: N/A. It doesn't build.

**Any additional details**

### Build log

    user@matter-vm:~/esp-matter/examples/generic_switch$ idf.py --preview set-target esp32c6                                       Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
    Executing action: fullclean
    Directory '/home/user/esp-matter/examples/generic_switch/build' doesn't seem to be a CMake build directory. R                  efusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
    user@matter-vm:~/esp-matter/examples/generic_switch$ idf.py build
    Executing action: all (aliases: build)
    Running cmake in directory /home/user/esp-matter/examples/generic_switch/build
    Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/user/.espressif/python_env/idf5.2_py3.10_env                  /bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/user/esp-matter/examples/generic_switch"...
    -- IDF_TARGET not set, using default target: esp32
    -- Found Git: /usr/bin/git (found version "2.34.1")
    -- Initialising new submodule components/bootloader/subproject/components/micro-ecc/micro-ecc...
    Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-                  ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc                  '...
    Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out '24c60e243580c78                  68f4334a1ba3123481fe1aa48'
    -- Initialising new submodule components/bt/controller/lib_esp32...
    Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for                   path 'components/bt/controller/lib_esp32'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/controller/lib_esp32'...
    Submodule path 'components/bt/controller/lib_esp32': checked out '943b0f24eb6ed390213599e4a9556c66844ef0db'
    -- Initialising new submodule components/bt/controller/lib_esp32c2/esp32c2-bt-lib...
    Submodule 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' (https://github.com/espressif/esp32c2-bt-lib.g                  it) registered for path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/controller/lib_esp32c2/esp32c2-bt-lib'...
    Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out 'ea33fbad1fa0879fe0e118359d2                  0463b3e2f126b'
    -- Initialising new submodule components/bt/controller/lib_esp32c3_family...
    Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) regi                  stered for path 'components/bt/controller/lib_esp32c3_family'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/controller/lib_esp32c3_family'...
    Submodule path 'components/bt/controller/lib_esp32c3_family': checked out 'af6be404ea583da57f9a4f0dfe7e02351ff                  5aa0d'
    -- Initialising new submodule components/bt/controller/lib_esp32c6/esp32c6-bt-lib...
    Submodule 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib' (https://github.com/espressif/esp32c6-bt-lib.g                  it) registered for path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib'...
    Submodule path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib': checked out 'a763b33ce1f6bdc257947845b05                  20c3b44de87eb'
    -- Initialising new submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib...
    Submodule 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib' (https://github.com/espressif/esp32h2-bt-lib.g                  it) registered for path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/controller/lib_esp32h2/esp32h2-bt-lib'...
    Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out 'c8018101d5a71162c220782e6c8                  60e9ba33013c4'
    -- Initialising new submodule components/bt/host/nimble/nimble...
    Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path                   'components/bt/host/nimble/nimble'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/bt/host/nimble/nimble'...
    Submodule path 'components/bt/host/nimble/nimble': checked out '98b88eacee2daf00f19c4ff52855cc3327d12e81'
    -- Initialising new submodule components/cmock/CMock...
    Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/cmock/CMock'...
    Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
    Submodule 'vendor/c_exception' (https://github.com/throwtheswitch/cexception.git) registered for path 'components/cmock/CMock/vendor/c_exception'
    Submodule 'vendor/unity' (https://github.com/throwtheswitch/unity.git) registered for path 'components/cmock/CMock/vendor/unity'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/cmock/CMock/vendor/c_exception'...
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/cmock/CMock/vendor/unity'...
    Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
    Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
    -- Initialising new submodule components/esp_coex/lib...
    Submodule 'components/esp_coex/lib' (https://github.com/espressif/esp-coex-lib.git) registered for path 'components/esp_coex/lib'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/esp_coex/lib'...
    Submodule path 'components/esp_coex/lib': checked out '68cbbdb45cdc8c88ffbfc2587a4e7af020695e56'
    -- Initialising new submodule components/esp_phy/lib...
    Submodule 'components/esp_phy/lib' (https://github.com/espressif/esp-phy-lib.git) registered for path 'components/esp_phy/lib'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/esp_phy/lib'...
[... 38 lines cut ...]
    Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/protobuf-c/protobuf-c'...
    Submodule path 'components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
    -- Initialising new submodule components/spiffs/spiffs...
    Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/spiffs/spiffs'...
    Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
    -- Initialising new submodule components/unity/unity...
    Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
    Cloning into '/home/user/esp-idf-v5.2/esp-idf/components/unity/unity'...
    Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
    -- Component directory /home/user/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/user/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/user/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
[... cut ...]
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:491 (__project_init)
      CMakeLists.txt:44 (project)
    
    
    -- Configuring incomplete, errors occurred!
    cmake failed with exit code 1, output of the command is in the /home/user/esp-matter/examples/generic_switch/build/log/idf_py_stderr_output_86601 and /home/user/esp-matter/examples/generic_switch/build/log/idf_py_stdout_output_86601
    user@matter-vm:~/esp-matter/examples/generic_switch$

sayondeep, 2023-11-16:
@lboue can you please try with idf commit id 6b1f40b9b as mentioned here : https://github.com/espressif/esp-matter#supported-esp-idf-and-connectedhomeip-versions].

lboue, 2023-11-16:
I switched to commit 6b1f40b9bf but I fails the same way

    git checkout 6b1f40b9bf

```
    user@matter-vm:~/esp-matter/examples/generic_switch$ idf.py --preview set-target esp32c6
    Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
    Executing action: fullclean
    Directory '/home/user/esp-matter/examples/generic_switch/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
    user@matter-vm:~/esp-matter/examples/generic_switch$ rm -rf build/*
    user@matter-vm:~/esp-matter/examples/generic_switch$ idf.py --preview set-target esp32c6
    Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
    Executing action: fullclean
    Build directory '/home/user/esp-matter/examples/generic_switch/build' is empty. Nothing to clean.
    Executing action: set-target
    Set Target to: esp32c6, new sdkconfig will be created.
    Running cmake in directory /home/user/esp-matter/examples/generic_switch/build
    Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/user/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=0 /home/user/esp-matter/examples/generic_switch"...
    -- Found Git: /usr/bin/git (found version "2.34.1")
    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
      submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/bt/controller/lib_esp32c2/esp32c2-bt-lib is out of
      date.  Run 'git submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/bt/controller/lib_esp32c3_family is out of date.
      Run 'git submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/bt/controller/lib_esp32c6/esp32c6-bt-lib is out of
      date.  Run 'git submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib is out of
      date.  Run 'git submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
[... 78 lines cut ...]

    CMake Warning at /home/user/esp-idf-v5.2/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
      Git submodule components/openthread/openthread is out of date.  Run 'git
      submodule update --init --recursive' to fix.
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:80 (git_submodule_check)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/build.cmake:240 (__build_get_idf_git_revision)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:29 (include)
      CMakeLists.txt:34 (include)

[... cut ...]
    Call Stack (most recent call first):
      /home/user/esp-idf-v5.2/esp-idf/tools/cmake/project.cmake:433 (__project_init)
      CMakeLists.txt:44 (project)

    -- Configuring incomplete, errors occurred!
    cmake failed with exit code 1, output of the command is in the /home/user/esp-matter/examples/generic_switch/build/log/idf_py_stderr_output_91395 and /home/user/esp-matter/examples/generic_switch/build/log/idf_py_stdout_output_91395
    user@matter-vm:~/esp-matter/examples/generic_switch$
```

sayondeep, 2023-11-17:
@lboue can you please confirm that you have followed all the steps mentioned here : https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories.

Also, you have updated the submodules for both esp-matter and esp-idf.

sayondeep, 2023-11-20:
@lboue following the instructions mentioned in the previous comment..the build for esp32c6 is working fine.

lboue, 2023-11-20:
Building is working after deleting everything and starting again. Thanks

closed 2023-11-20 by lboue