espressif/esp-matter #736: OTA Rollback for Matter device

PhLuReh, 2023-11-15:
**Describe the question/query that you have**
What is the proposed way to perform an OTA rollback on our Matter devices?
How do we configure a factory image partition? Can we use the idf `menuconfig` way?

**Additional context**
I know the resources about [Matter OTA](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#matter-ota) and I also could perform an update using the chip-tool. Further I know how [OTA works](https://blog.espressif.com/ota-updates-framework-ab5438e30c12) and the [IDF commands](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html). Even a small topic discusses [software factory reset](https://www.esp32.com/viewtopic.php?t=31090).

sayondeep, 2023-11-17:
@PhLuReh From the Matter 1.2 Core Specification [here](https://csa-iot.org/developer-resource/specifications-download-request/). Please refer to **11.19.2. Functional overview** 
which states that "OTA Requestors SHALL only upgrade to numerically newer versions and OTA Providers SHALL only ever provide a numerically newer Software Image than a Node’s current version number. Any functional rollback SHALL be affected by the Vendor creating a Software Image with a newer (higher) version number, but whose binary contents may match prior func­tionality."

PhLuReh, 2023-11-17:
Thank you for stating this also with a reference. I already read this, but didn't get the point, that the user may also not be able to roll the system back. I thought the "security anti-rollback" should deal with this requirement.

Isn't the tester at the ATL requiring such a feature?

closed 2023-11-17 by PhLuReh