espressif/esp-matter #737: Water Content Measurement

goronlant, 2023-11-16:
**Describe the question/query that you have**
Due to my personal project, I want to develop a soil moisture device for my plants. In the document "Matter 1.1 Application Cluster Specification" under 2.6. Water Content Measurement, it states that there should be a cluster called Soil Moisture Measurement with the identifier 0x0408. The issue is that it is not found anywhere in the code. Does the feature exist, or is it still to come?

If there are alternative ways to implement this, I would greatly appreciate it if you could provide that information.

Thanks in advance.

**Additional context**

I am uploading a screenshot from the document. 
[image]

PSONALl, 2023-11-16:
@goronlant This cluster is not yet implemented in esp-matter
I will add it soon

goronlant, 2023-11-17:
Okay, thanks! Is it possible to implement it as a non-esp developer?

goronlant, 2023-11-17:
Yeah, I've also raised that issue. This isn't ideal news for me since there's currently no alternative solution on the market besides Matter, which is compatible with all smart devices as well. Do you know if it is possible to implement this feature as a third-party provider?

dhrishi, 2023-11-22:
I will close this here. You can track https://github.com/project-chip/connectedhomeip/issues/30508

closed 2023-11-22 by dhrishi