espressif/esp-matter #738: strnchr in rainmaker.cpp

jonsmirl, 2023-11-17:
These strchr need to be strnchr. This string is not zero terminated. You have size a couple lines above. 
~~~
    /* The expected format of the data is "<user_id>::<secret_key>" */
    char ch = ':';
    char *check_first = strchr(data, (int)ch);
    char *check_second = NULL;
    if (check_first && (size >= (int)((check_first + 1) - data))) {
        check_second = strchr(check_first + 1, (int)ch);
~~~
https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L259

shripad621git, 2023-11-22:
@jonsmirl, thanks for the suggested change.We will fix it .

shubhamdp, 2023-12-08:
This was fixed in e4b0c71eab982fd42e7d2a57cef2a46d8753f02e and 589091f4c876bde7af02ec7642034a38a2beaa65

closed 2023-12-08 by dhrishi