espressif/esp-matter #739: How to Change Name or Set Name of Matter device in Espressif Mattter SDK?

bilalmalik76, 2023-11-18:
We want to change the name or set the name of our matter device. For example, if we are using light example available in Espressif Mattter SDK? and we want to change it name to something like company_name_light? How can we do it in Espressif Mattter SDK?

jonsmirl, 2023-11-18:
https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool

wqx6, 2023-11-20:
Do you mean the name shown in iOS Apple home app when pairing the device? The Apple Home app will fetch the device name and device type from DCL MainNet. So you should firstly obtain the certificate and register your device on MainNet.

dhrishi, 2023-11-22:
@bilalmalik76 Please provide the required inputs

bilalmalik76, 2023-11-22:
@wqx6 Yes the name shown in iOS Apple home app when pairing the device.

dhrishi, 2023-11-22:
> The Apple Home app will fetch the device name and device type from DCL MainNet. So you should firstly obtain the certificate and register your device on MainNet.

Then, the above is the answer. You need to certify your produce, and once done update the DCL with these details

closed 2023-11-22 by dhrishi