espressif/esp-matter #740: ESP32-C6 as Open Thread Border Router

bilalmalik76, 2023-11-18:
Considering ESP32-C6 has both Wi-Fi and Thread radio in a single chip. Can we use it as a Thread border router? If yes, there is any guide to setup ESP32-C6 as a border router?

jonsmirl, 2023-11-18:
For a border router you need to add ESP32-H type chip.
https://docs.espressif.com/projects/esp-thread-br/en/latest/
https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html

This has to do with the C6 802.11 radio being shared and to be a border router you have to listen 100% of the time.  So adding the H2 allows you to listen 100% of the time.

There is really not much need to make a border router. All of the Google, Apple, Amazon Home devices function as a border router.

bilalmalik76, 2023-11-19:
@jonsmirl what's the point of having both Wi-Fi and Thread radio in single SoC then?

jonsmirl, 2023-11-19:
You can make an end device with the C6 that can pick wifi or thread networking.

jonsmirl, 2023-11-19:
You can also make a controller with it:
https://github.com/espressif/esp-matter/tree/main/examples/controller

bilalmalik76, 2023-11-19:
So ESP32-C6 can act both as a matter controller and Matter over Thread device at a time?

jonsmirl, 2023-11-19:
I don't know about at the same time, have to ask the Espressif people.

dhrishi, 2023-11-22:
Closing as the question is answered

closed 2023-11-22 by dhrishi

BohdanK-W32, 2025-03-23:
> > Can we use it as a Thread border router? If yes, there is any guide to setup ESP32-C6 as a border router?
> 
> Sure, you can use ESP32-C6 as a Thread border router, but it is not recommended as there is only one RF path in ESP32-C6. We have a [DevKit board](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#id2) for Thread BR
> 
> > So ESP32-C6 can act both as a matter controller and Matter over Thread device at a time?
> 
> Yes, but normally the Matter controller is a Matter-over-Wi-Fi device. We are planning to integrate Thread BR and the [matter controller example](https://github.com/espressif/esp-matter/tree/main/examples/controller), and the code is under internal review.

Was the example implemented somewhere?