espressif/esp-matter #741: Override Callback doesn't provide endpoint priv_data

germouse12, 2023-11-19:
Why doesn't the execute_override_callback function include the priv_data in the call like it does for the normal callback? 

https://github.com/espressif/esp-matter/blob/d17b310177bbee3cf88197d54077c3458413085f/components/esp_matter/esp_matter_attribute_utils.cpp#L1002

vs

https://github.com/espressif/esp-matter/blob/d17b310177bbee3cf88197d54077c3458413085f/components/esp_matter/esp_matter_attribute_utils.cpp#L992

This difference is not documented, so I had to work through some errors before I realized it wasn't passing the priv_data (which holds a pointer to my driver in my case).

shubhamdp, 2023-11-21:
@germouse12 Looks like a bug, will fix it.

germouse12, 2023-12-05:
Just realized I closed this before asking if you wanted me to keep it open to address the above identified issue. Or I can open a different issue? Let me know what me to do.

dhrishi, 2023-12-05:
@germouse12 Keeping this open is fine. Thanks!

closed 2024-01-23 by espressif-bot