espressif/esp-matter #742: How does the override callback functionality work?

germouse12, 2023-11-19:
I am working on a project where the state of the device can change without the esp's awareness. Instead of frequently polling the device, I would like to use this functionality to only poll it when the information is requested.

Do I need to do anything more than call attribute::set_override_callback? 

Do I need to notify the system if the attribute has been updated externally (which would likely require me to poll the device anyway)?

Why do we need to provide a callback for overriding when the endpoint's attribute update callback could handle this function using the READ and WRITE types? Is there a way to get these types and use the normal callback without running set_override_callback?

jonsmirl, 2023-11-19:
Most controller apps don't ask for the state of attributes, instead they do a wildcard subscribe on the device.  Then as the attributes change they get report notifications. The majority of apps (everyone I have looked at) use subscriptions to monitor the device state.  So it looks like you will have to do polling unless you can get some kind of notification from the external device.

shubhamdp, 2023-11-21:
@germouse12 Please take a look at the documentation, according to this, if you have an attribute which needs special handling when reading/writing the attribute then set this callback and perform the read/write operations in there.

https://github.com/espressif/esp-matter/blob/d26a261266ecbe6fff37e6693b4df3f588fab4a2/components/esp_matter/esp_matter_core.h#L558-L573

shubhamdp, 2023-12-03:
@germouse12 please close the issue of question is answered.

germouse12, 2023-12-05:
I will close, but I am still not clear on how to use this function. If I call report or update or similar functions will it also call the override callback with the READ or WRITE type? I guess I am still not entirely sure what the framework manages and what I have to manage.

I think I have gotten the answer to my second question, I will have to do my own polling.

closed 2023-12-05 by germouse12