espressif/esp-matter #744: ESP Matter Genric Switch to Control LED on ESP Matter Light Example

bilalmalik76, 2023-11-19:
Can I use the ESP Matter Genric Switch example on my one esp32 to control LED on another ESP32 using ESP Matter light example?

jonsmirl, 2023-11-19:
No, generic switch only sends event back to a controller like Apple Google Amazon Home.

bilalmalik76, 2023-11-19:
@jonsmirl but controller will ultimately send this response to a device to which generic switch is attached right?

bilalmalik76, 2023-11-19:
@jonsmirl In the Apple Home app, we can define actions and link a generic switch to a specific accessory. I tried with one esp32 light example and another with esp32 generic switch example. After linking them together in apple home app actions, genetric switch esp32 is able to turn off LED on esp32 light but it is not able to turn it off. That means controller is sending generic switch events to esp32 light, but I am not sure why not able to turn on light, but turning off works fine.

jonsmirl, 2023-11-19:
The Generic Switch is sending Generic Switch Events to Apple Home. Apple Home then receives those events and sends OnOff commands in response.   You probably want to be building a controller.  https://github.com/espressif/esp-matter/tree/main/examples/controller

bilalmalik76, 2023-11-19:
@jonsmirl If Apple Home sends onOff in response, then LED on ESP32 light should also turn on and off accroding to OnOff cluster value, isn't it?

jonsmirl, 2023-11-19:
There are two commands -- On and Off.  You are likely only sending the On command.

bilalmalik76, 2023-11-20:
@jonsmirl Generic switch send events based on  switch_cluster::event::send_initial_press() and witch_cluster::event::send_short_release() and Apple Home send response to linked devices in response. But in my case only On is being sent either I use send_initial_press() or send_short_release(). Can anyone from espressif help me resolve this issue?

bilalmalik76, 2023-11-20:
If you see this [(](https://www.youtube.com/watch?v=AJ1K829B0S4&t=384s) tuo smart button, it is based on a matter generic switch and it is able to control lights. I think there is something missing in esp generic switch example.

dhrishi, 2023-11-20:
> Can I use the ESP Matter Genric Switch example on my one esp32 to control LED on another ESP32 using ESP Matter light example?

No, you cannot. Generic switch device type has switch events like single press, multi press, long press etc. that can be sent across to the controllers. It does not have a OnOff client cluster through which it can interact with a light having OnOff server cluster.

>  But in my case only On is being sent either I use send_initial_press() or send_short_release(). 

If you are doing the automation through the ecosystem app, can you confirm if you are doing it right. Matter is not involved in this part. I am assuming, it will be something like - When you get event X/Y from the generic switch, turn on/off the light

jonsmirl, 2023-11-20:
> If you are doing the automation through the ecosystem app, can you confirm if you are doing it right. Matter is not involved in this part. I am assuming, it will be something like - When you get event X/Y from the generic switch, turn on/off the light

That is what Apple Home is doing.

bilalmalik76, 2023-11-20:
Yes I am using Automation in Apple home app. I tested with Tuo Button with
same Automation settings and able to turn on and off light. But there is
some issue with ESP32 Generic switch example, it only sends eithet event x
or y, therefore, it only able to turn on led but not able to turn it on.

jonsmirl, 2023-11-20:
The Tuo Button might be implementing a two state switch. The generic switch example is implementing a single state switch. You'd need to add a little more code to the generic switch example to do a two state switch.

Use chip-tool to see how Tuo is sending events.

bilalmalik76, 2023-11-20:
@jonsmirl Apple home app does not support latch type switch

jonsmirl, 2023-11-20:
Then you need to use chip-tool to see what switch events the Tuo switch is sending.

bilalmalik76, 2023-11-20:
Yes doing it now.

dhrishi, 2023-11-22:
I am closing this issue as the original question is answered

closed 2023-11-22 by dhrishi