espressif/esp-matter #746: generic_switch matter onboardingcodes qrcode  ==> Error: 47

lboue, 2023-11-20:
**Describe the bug**
I can't generate QR code for pairing the device

**Environment**
- ESP-Matter Commit Id: 01bebca56b42a35b5ad049f59292df7d4c08ea6b
- ESP-IDF Commit Id: d26a261266ecbe6fff37e6693b4df3f588fab4a2
- SoC: ESP32-C6
- Device Logs 
- Host Machine OS: Ubuntu 22.04.2 LTS (GNU/Linux 6.2.0-36-generic x86_64
- Commissioner app and versions if present: N/A. It doesn't build.
- Commissioner's logs if present: N/A. It doesn't build.

**Any additional details**

    /home/user/esp-idf-v5.2/esp-idf/export.sh
    cd ~/esp-matter/examples/generic_switch
    idf.py --preview set-target esp32c6
    idf.py build
**Any additional details**
...

```
> matter onboardingcodes qrcode

Error: 47
```

shubhamdp, 2023-11-21:
@lboue This is expected... please take a look at https://github.com/espressif/esp-matter/issues/658#issuecomment-1733340465, we do not have setup passcode in the firmware hence we cannot generate the qrcode.

jonsmirl, 2023-11-22:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control

The dev qrcode is in section 2.2.1.1

dhrishi, 2023-11-22:
@lboue In addition to the above replies, if you are specifically looking for the generic_switch example QR code, you can see it in the example README: https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/README.md#1-additional-environment-setup

closed 2023-11-22 by dhrishi

lboue, 2023-11-22:
> @lboue In addition to the above replies, if you are specifically looking for the generic_switch example QR code, you can see it in the example README: https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/README.md#1-additional-environment-setup

@dhrishi It doesn't work for ESP-C6