espressif/esp-matter #751: There is a problem with the 'light_switch' example

mohankrishkumare, 2023-11-22:
**Describe the bug**
There is a problem with the 'light_switch' example. It shows 'Not Supported' in Apple's Home app after adding the accessory.
[image]

But, the 'light' example is working fine.

**Environment**
- ESP-Matter Commit Id: 79dfcc7b9db37d80105e2bbb39a680a0670296d5
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file): [log.txt](https://github.com/espressif/esp-matter/files/13440669/log.txt)
- Host Machine OS: Ubuntu 22.04.3 LTS
- Commissioner app and versions if present: 
- Commissioner's logs if present: 

**Any additional details**
...

mohankrishkumare, 2023-11-22:
iPadOS Version: 16.7.2

dhrishi, 2023-11-22:
@mohankrishkumare Light switch, that has a cluster client, is not supported in the Ecosystem apps. It can commission the device, but cannot recognize the device type. As a result, what you are seeing is expected.

closed 2023-11-22 by dhrishi

mohankrishkumare, 2023-11-22:
How to implement the 'Retro Fit Switch' (GPIO) in ESP ZeroCode?