espressif/esp-matter #752: matter-bridge  Accessories are not displayed in Apple Home

WOLEN6914183, 2023-11-23:
I currently have a product implemented based on the official example/matter-bridge-app. My ZAP file structure has endpoint 0 as the root node, endpoint 1 as an Aggregator device type, endpoint 2 as a Thermostat device type, and endpoint 3 as a Matter Dimmable Light. Endpoint 3 serves as a bridge node, bridging four lights below it, while endpoint 2 functions as a regular Matter device (note: this configuration is allowed in the official specifications). After configuring the devices in the Apple Home app, I noticed that only the accessories for the four lights under the bridge node appeared, and the thermostat under endpoint 2 did not. When I read the properties of the thermostat under endpoint 2 using a Raspberry Pi, I can retrieve the information. Moreover, I can see the thermostat accessory in the Google Home app. Is this an issue of incompatibility with Apple HomeKit, or is there another explanation for this behavior?

wqx6, 2023-11-23:
[image]
 
The Bridged endpoints should be in the Part List of the aggregator endpoint so that it can be identified as a bridged device by eco-system of Apple or Google. we provide `set_parent_endpoint()` in esp-matter the set parent for bridged endpoints.

dhrishi, 2023-11-27:
@WOLEN6914183 Can you please check if this helps?

WOLEN6914183, 2023-11-27:
[image]

"I have reviewed the official documentation, and our product is similar to the structure in the diagram above. On the bridge device, there is a thermostat similar to the 'ep31' in the diagram, but it does not belong to the 'Part List of the aggregator' node. I can see the thermostat accessory in the Google Home, but it is not visible in the Apple Home."

wqx6, 2023-11-27:
If the thermostat endpoint is not child endpoint of the aggregator endpoint(not in the PartList of aggregator endpoint), it should be visible as an independent (non-bridged) endpoint in Apple home app. If you cannot find the device in Apple Home app, I think it is Apple home app's issue.

WOLEN6914183, 2023-11-27:
ok, thank you for your answer despite your busy schedule. I will ask Apple's customers when I have time.

closed 2023-11-27 by WOLEN6914183