espressif/esp-matter #753: Can the sleep_device demo be woken by GPIO?

KEEPHEADU, 2023-11-23:
**Describe the question/query that you have**
I want to know whether the sleep_device demo belongs to light-sleep or deep-sleep mode, and whether the sleep mode it is in will be woked up by GPIO. If it can't be woked up by GPIO, Can I set GPIO wakeup using gpio_wakeup_enable()?

**Additional context**

kelin6, 2023-11-23:
@KEEPHEADU 

Sleep_device demo belongs to light-sleep.   

ESP32H2 GPIO power domain belong to  the TOP power domain, in light sleep, the TOP power domain is powered down, so waking up through GPIO is not possible and also cannot set GPIO wakeup using `gpio_wakeup_enable()`. However, waking up through `RTC IO` is possible and you can using `esp_sleep_enable_ext1_wakeup_with_level_mask`,  please referred to [RTC IO Wakeup](https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_sleepy_device/deep_sleep/main/esp_ot_sleepy_device.c#L154) and  [ext1_wakeup](https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/ext_wakeup.c#L45C21-L45C65).

KEEPHEADU, 2023-11-23:
@kelin6 If the sleep_device is light_sleep then it should be possible to use GPIO to wake up the sleep_device. This is some information I saw.[https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32h2/api-reference/system/sleep_modes.html?highlight=sleep#gpio](url)

KEEPHEADU, 2023-11-24:
@kelin6  Thank you for your answer, I have a new problem now, I need to use the serial port in the sleep_device demo, but the use of serial port in the light_sleep process does not work as expected, I want to know if there is any way to keep the chip active? I thought about using the StayActiveRequest command from ICD Management Cluster, but I don't know if it's actually implemented, and I didn't find a way to use it.

wqx6, 2023-11-24:
`StayActiveRequest` command in ICD management cluster will only make the device work with a fast polling interval(Which is Active mode for ICD) for a duration. The chip will still sleep after receiving this command. But the chip will wake up more frequently than Idle mode. If you want the device keep waking up, this command won't work. 

You can use `esp_matter::cluster::icd_management::command::create_stay_active_request()` to create this command.

Currently, this command should include a command data field `StayActiveDuration` in SPEC, but this command data field is not supported in connectedhomeip repo. So the ActiveDuration will always be the default duration which can be configured in menuconfig

KEEPHEADU, 2023-11-24:
@kelin6  Yes, when I perform `esp_matter: before using uart: cluster: : icd_management: : command: : create_stay_active_request ()` still can't normal to send and receive data through uart, How can I send and receive data using uart normally during light sleep?

KEEPHEADU, 2023-11-24:
[image]
In the function block diagram, the power off of UART is optional. Can I configure the power off of UART to make UART work normally during light sleep? How do I configure it?

kelin6, 2023-11-24:
@KEEPHEADU 
It is important to note that the power domain for UART is TOP, and the power domain for UART is not independent. This needs to be explicitly mentioned in this document. I have already communicated this internally. In the future, this point should be clarified in the ESP32H2 documentation.

Based on your requirements, you can achieve this by using `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=n` to keep TOP powered on, and then use [ UART wake-up](https://github.com/espressif/esp-idf/blob/master/examples/system/light_sleep/main/uart_wakeup.c) needed.

kelin6, 2023-11-27:
@KEEPHEADU In light of your previous requirements, are you aiming to achieve the following functionality: The host connects to ESP32H2, and the host wakes up ESP32H2 through GPIO. After ESP32H2 wakes up, it communicates with the host using UART for data transmission and reception?

KEEPHEADU, 2023-11-27:
@kelin6 Yes, the sensor wakes up the ESP32H2 via GPIO, and after the matter network wakes up, it communicates with the sensor via UART.

The problem now is that after the matter network wakes up, UART communication is not possible.

kelin6, 2023-11-29:
@KEEPHEADU  Since UART0 is used for debug log output, it is recommended to use UART1 for communication after waking up. Also, set CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP to n. I have tested in the OpenThread light sleep example and confirmed that data can be sent and received using UART1 after waking up.

The  code used is as follows, you can refer to it.

```
#include "driver/uart.h"
#define UART_NUM1       (UART_NUM_1)
#define UART1_TX_PIN    (1)
#define UART1_RX_PIN    (2)
static void test_uart1_init()
{
    const int uart_num = UART_NUM1;

    uint8_t tx_pin = UART1_TX_PIN;
    uint8_t rx_pin = UART1_RX_PIN;
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,
        .source_clk = UART_SCLK_DEFAULT,
        .rx_flow_ctrl_thresh = 120
    };
    uart_driver_install(uart_num, 1024 * 2, 0, 20, NULL, 0);
    uart_param_config(uart_num, &uart_config);
    uart_set_pin(uart_num, tx_pin, rx_pin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}

static void test_uart1_tx()
{
    printf("Send data start\n");
    int packlen = 100;
    uint8_t *tx_buf = (uint8_t *)malloc(packlen);
    if(tx_buf == NULL) {
        ESP_LOGE("","tx buffer malloc fail");
    }
    for(int i = 0; i < packlen; i++) {
        tx_buf[i] = i;
    }
    uart_write_bytes(UART_NUM1, (const char*)tx_buf, packlen);
    uart_wait_tx_done(UART_NUM1, portMAX_DELAY);
    free(tx_buf);
    printf("Send data with break test passed\n");
}

static void test_uart1_rx()
{   int len = -1;
    uint8_t buf_size = 255;
    uint8_t *data = (uint8_t *) malloc(buf_size);
    if(data == NULL) {
        ESP_LOGE("","read buffer malloc fail");
    }
    int64_t start_time = esp_timer_get_time();
    while ((esp_timer_get_time()-start_time)< 100000) {
        len = uart_read_bytes(UART_NUM1, data, buf_size, 0);
        if (len < 0) {
            ESP_LOGI("read timeout", "uart read write test fail");
            break;
        } else if (len > 0) {
            printf("rx len:%d, 0x%x 0x%x\n", len, data[0], data[1]);
        }    
    }
    free(data);
}
```

wqx6, 2023-11-29:
> @kelin6 Yes, the sensor wakes up the ESP32H2 via GPIO, and after the matter network wakes up, it communicates with the sensor via UART.
> 
> The problem now is that after the matter network wakes up, UART communication is not possible.

What do you mean by `matter network wakes up`? The matter network will not sleep or wake up. There are only two mode for a sleepy end device, which are Idle mode and active mode. The only difference of the two mode is the frequency of the radio-wake-up. If  you want to make the soc keep waking up for a while and do the UART communication during that time, there is nothing to do on matter stack. You should just do something like: wake up the soc with the GPIO, and after the soc wakes up, notify the UART task to process the work.

KEEPHEADU, 2023-11-30:
@wqx6 Yes, I should notify the uart task to work after waking up.

But when I use uart_read_bytes() to receive the sensor data, it doesn't immediately receive the correct ack response data from the sensor, the first few times it is good, but the subsequent data is not.

Is it somewhere in the sleep_device demo in esp-matter that periodically calls esp_light_sleep_start() to enter light_sleep, causing uart to fail to receive data or to receive data incorrectly?

When I use the function to see why I wake up, sometimes it's "timer".

I'm really troubled.

KEEPHEADU, 2023-12-01:
@kelin6 @wqx6 Thank you for your answer, this problem has been solved, I use `esp_pm_lock_create` to register a lock `ESP_PM_NO_LIGHT_SLEEP`, when I use uart I should use `esp_pm_lock_acquire` the lock, and release it after using uart, Because the uart1 I'm using doesn't save the context.

closed 2023-12-01 by KEEPHEADU