espressif/esp-matter #755: How to use ESP Matter APP to connect with matter bridge (con 1 reason 0x213)

GeeklinkSmart, 2023-11-28:
Now I can use chiptool to connect with the matter bridge and control the light.
But when I use ESP Matter APP to connect with matter bridge ,I scan the QR code ,this is the logs.APP shows timeout after a while

I (28088) chip[DL]: BLE GAP connection established (con 1)
I (28089) chip[DL]: CHIPoBLE advertising stopped
I (29268) chip[DL]: BLE GAP connection terminated (con 1 reason 0x213)
I (29269) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (08:00:27.400) NimBLE: GAP procedure initiated: advertise; 
I (08:00:27.407) NimBLE: disc_mode=2
I (08:00:27.411) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (08:00:27.422) NimBLE: 

I (29297) chip[DL]: CHIPoBLE advertising started
I (32586) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (32587) chip[DL]: Device already advertising, stop active advertisement and restart
I (08:00:30.720) NimBLE: GAP procedure initiated: stop advertising.

I (08:00:30.728) NimBLE: GAP procedure initiated: advertise; 
I (08:00:30.734) NimBLE: disc_mode=2
I (08:00:30.738) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (08:00:30.749) NimBLE:

dhrishi, 2023-11-28:
@GeeklinkSmart The ESP Matter app was added as a test app to perform the commissioning, and is not maintained currently. Instead, the ESP RainMaker app is the default app for Matter from Espressif. The iOS app is already updated on the app store to support Matter while the Android app will be updated soon.

AFAIK, the bridge device type isn't supported in the app yet.

GeeklinkSmart, 2023-11-30:
@dhrishi now I can use IOS app Home to add it as matter accessory, and the bridge show online ,but the light shows no responed, So I can't control the light in Home app.
here is some logs. 
`I (4578063) chip[IN]: (S) Sending msg 224702362 on secure session with LSID: 17514
I (4580027) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (4580029) chip[EM]: <<< [E:39069i S:17514 M:224702363] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4580035) chip[IN]: (S) Sending msg 224702363 on secure session with LSID: 17514
I (4580044) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (4580322) chip[EM]: >>> [E:39069i S:17514 M:201822422 (Ack:224702363)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:01 (IM:StatusResponse)
I (4580325) chip[IM]: Received status response, status is 0x00
I (4580332) chip[EM]: <<< [E:39069i S:17514 M:224702364 (Ack:201822422)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4580346) chip[IN]: (S) Sending msg 224702364 on secure session with LSID: 17514
I (4582052) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (4582054) chip[EM]: <<< [E:39070i S:17514 M:224702365] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4582060) chip[IN]: (S) Sending msg 224702365 on secure session with LSID: 17514
I (4582069) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (4582106) chip[EM]: >>> [E:39070i S:17514 M:201822423 (Ack:224702365)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:01 (IM:StatusResponse)
I (4582109) chip[IM]: Received status response, status is 0x00
I (4582116) chip[EM]: <<< [E:39070i S:17514 M:224702366 (Ack:201822423)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4582130) chip[IN]: (S) Sending msg 224702366 on secure session with LSID: 17514
I (4583704) chip[EM]: >>> [E:46045r S:17514 M:201822424] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583707) chip[EM]: >>> [E:46046r S:17514 M:201822425] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583719) chip[DMG]: AccessControl: denied
I (4583724) chip[EM]: <<< [E:46045r S:17514 M:224702367 (Ack:201822424)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583737) chip[IN]: (S) Sending msg 224702367 on secure session with LSID: 17514
I (4583746) chip[DMG]: AccessControl: denied
I (4583751) chip[EM]: <<< [E:46046r S:17514 M:224702368 (Ack:201822425)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583764) chip[IN]: (S) Sending msg 224702368 on secure session with LSID: 17514
I (4583773) chip[EM]: >>> [E:46045r S:17514 M:201822426 (Ack:224702367)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4583788) chip[EM]: >>> [E:46047r S:17514 M:201822427] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583800) chip[EM]: >>> [E:46046r S:17514 M:201822428 (Ack:224702368)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4583815) chip[EM]: >>> [E:46048r S:17514 M:201822429] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583828) chip[DMG]: AccessControl: denied
I (4583832) chip[EM]: <<< [E:46047r S:17514 M:224702369 (Ack:201822427)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583845) chip[IN]: (S) Sending msg 224702369 on secure session with LSID: 17514
I (4583854) chip[DMG]: AccessControl: denied
I (4583859) chip[EM]: <<< [E:46048r S:17514 M:224702370 (Ack:201822429)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583872) chip[IN]: (S) Sending msg 224702370 on secure session with LSID: 17514
I (4583881) chip[EM]: >>> [E:46047r S:17514 M:201822430 (Ack:224702369)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4583896) chip[EM]: >>> [E:46049r S:17514 M:201822431] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583909) chip[EM]: >>> [E:46048r S:17514 M:201822432 (Ack:224702370)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4583923) chip[DMG]: AccessControl: denied
I (4583928) chip[EM]: <<< [E:46049r S:17514 M:224702371 (Ack:201822431)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583941) chip[IN]: (S) Sending msg 224702371 on secure session with LSID: 17514
I (4583962) chip[EM]: >>> [E:46049r S:17514 M:201822433 (Ack:224702371)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (4583967) chip[EM]: >>> [E:46050r S:17514 M:201822434] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0001:02 (IM:ReadRequest)
I (4583980) chip[DMG]: AccessControl: denied
I (4583984) chip[EM]: <<< [E:46050r S:17514 M:224702372 (Ack:201822434)] (S) Msg TX to 1:00000000B8C4A5C0 [ED3C] --- Type 0001:05 (IM:ReportData)
I (4583997) chip[IN]: (S) Sending msg 224702372 on secure session with LSID: 17514
I (4584014) chip[EM]: >>> [E:46050r S:17514 M:201822435 (Ack:224702372)] (S) Msg RX from 1:00000000B8C4A5C0 [ED3C] --- Type 0000:10 (SecureChannel:StandaloneAck)`

GeeklinkSmart, 2023-11-30:
[image]

dhrishi, 2023-11-30:
@GeeklinkSmart Does the default BLE Mesh bridge example not work with the Home app? If yes, please create a new issue and provide details about the iOS version. Also, mention if you are doing any changes in your app. I will close this one as the original question is answered.

closed 2023-11-30 by dhrishi

JetForMe, 2025-01-28:
> The ESP Matter app was added as a test app to perform the commissioning, and is not maintained currently. Instead, the ESP RainMaker app is the default app for Matter from Espressif.

You guys should update the instructions in [Section 2.3 here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control), as it still points the reader to the old app, which no longer exists in the App Store.

shubhamdp, 2025-01-28:
@JetForMe Thanks for pointing out, will update the links!