espressif/esp-matter #756: Release-based docker images

PhLuReh, 2023-11-28:
**Is your feature request related to a problem? Please describe.**
As reported in https://github.com/espressif/esp-matter/issues/747, I'm using your [docker image](https://hub.docker.com/r/espressif/esp-matter) to have a reproduceable build framework. On my framework I'm using a sshd to be able to use VSCode Remote.

When changing the branch of esp-matter, I ran into different problems with git unable to change branches due to heavy usage of submodules.

**Describe the solution you'd like**
I would like to have Matter release specific docker images. While searching the repos, I found, that the [build-file](https://github.com/espressif/esp-matter/blob/main/.github/workflows/docker-image.yml) of your docker-image already can handle branch-specific image creation. It would be very beneficial to have those, ready for production usage.

**Describe alternatives you've considered**
As described, changing the branch of the [provided images](https://hub.docker.com/layers/espressif/esp-matter/latest_idf_v5.1.1/images/sha256-0b101e9ef05000721debe2071f435e723788ea81e1612f22ae0ed283b9aaa425?context=explore) end into having trouble and possibly inconsistent build environments.

**Additional context**

shubhamdp, 2024-02-09:
We have started preserving release based docker image and you can find it here: https://hub.docker.com/r/espressif/esp-matter/tags.

Although we do not have one for release-v1.1

PhLuReh, 2024-02-12:
nice to hear, good job! unfortunately, I'm currently running on v1.1, so I will have to wait for the next release of our products... hopefully based on at least v1.2

closed 2024-02-12 by PhLuReh

shubhamdp, 2024-02-12:
@PhLuReh We will backport the change which builds and pushes the docker image on release/v1.1 and you'll have an image for v1.1 as well.